XCSoar/XCSoar #462: Sync time with GPS causes invalid time stamps in IGC

dingetje, 2020-07-30:
XCSoar versions having and not having the problem
-------------------------------------------------
Detected using XCSoar 6.8.16, not sure what version does *not* show the behavior.

Expected behavior
-----------------
Sync once with GPS at start of flight and then use internal clock to avoid time "jumps"

Actual behavior
---------------
Time jumps in B-records of IGC, causing log to be aborted on the OLC

Steps to reproduce the behavior
-------------------------------
- enable sync time with GPS
- go fly
- upload IGC file on OLC

Do you have any idea what may have caused this?
-----------------------------------------------
A resync of the internal clock with GPS time may cause time "jumps" no longer accepted on the OLC

Do you have an idea how to solve the issue?
-------------------------------------------
Sync once at start of log with GPS clock and switch to internal clock for the remainder of the flight to avoid time "jumps"

MaxKellermann, 2020-08-02:
Which operating system is this?

lordfolken, 2020-09-17:
@dingetje any updates on this?

dingetje, 2020-09-23:
Sorry for the delay. Here's the IGC file:

[2020-07-29-XCS-AAA-01.zip](https://github.com/XCSoar/XCSoar/files/5266115/2020-07-29-XCS-AAA-01.zip)

Here's the OLC validation error message:

> 200729 162737 - A recording gap over 60 seconds (record: "B114721...") was found that does not conform to the rules. The flight was shortened from there.
200729 162737 - File Quality is 20%

August2111, 2021-01-05:
This issue has nothing to do with operation system or something like this. This problem happens every time if the internal clock of the system a little bit async with the GPS time - and the system synchronize during the recording of the flight. If the time jump forward but smaller the 60s - nothing happens, the flight is ok in the OLC analysis. If you have a forward jump over 60s there is the same problem like a reboot of the system with a gap of more than 60s: The OLC will shortened the flight on this point... If yoe have a jump backward, the OLC says this is completely forbidden and will stopped the analysis also on this point! If you have this behaviour before the take off is happend - the complete flight file is useless ;-( even thought the complete real flight is recorded correctly.
This was a very new strong behaviour in 2020, nobody had this problem before!
BTW.: This issue also happens on Flarm, LXNav and and all other IGC file systems with synchronizing during the recording!

lordfolken, 2021-04-08:
@teji24 does this happen with the "use gps" time option enabled?

lordfolken, 2021-07-05:
If your clock is that unstable, please use the gps time option or change your hardware.

closed 2021-07-05 by lordfolken

August2111, 2021-07-05:
From my point of view this should not be closed, because this is a normal behaviour! If the clock ist to fast - only less seconds (1 second is enough!) after a 10h flight a sync of the the gps time destroyed the usage of this igc file on OLC!
Ok, you can say: 'OLC is out of the date' - but this isn't really true and the time base of a flight should not change during a flight!
And much more critical is the case changing of time base after logger switched on and later the 1st connection to GPS - and this is recorded in the IGC file... ;-(