XCSoar/XCSoar #464: Fraction for ballast in percent

iltis42, 2020-08-14:
Put 160 Liters for my Nimbus 2, what is the maximum water ballast. 
Given from Parser.cpp a setting in percent of capacity, i expected a value of !g,b100,
exactly looking at: 
<12> Instrument Ballast setting in percent of capacity

Hence what my Vario receives is "**!g,b10**", i am missing a zero at the end:
Received RFCOMM data packet on channel id 4, size 7 !g,b10
   67 2c 62 31 30 0d 
               b   1   0  

Same for smaller fractions, e.g. 80 Liters or 50% give just a '5' (%):
Received RFCOMM data packet on channel id 4, size 6 !g,b5
   67 2c 62 35 0d 
               b  5  

Looking at the implementation, i guess  'fraction * 100' was needed to get percentage, 
CAI302::PutBallast(Port &port, double fraction, OperationEnvironment &env)
{
  unsigned ballast = uround(fraction * 10);

Can you confirm this finding and maybe fix for the future release's ?

Thanks in advance !

iltis42, 2020-08-20:
Any news about here ?

lordfolken, 2020-08-20:
This change from 100 to 10 was done in 2012 to fix some issue 32078a963fc @MaxKellermann 
I'm somewhat suspicous if that didn't catch someones attention in the meantime.
Please create a fix and test it. Please also lobby with borgelt to keep a copy of the protocol publicly available.

iltis42, 2020-08-21:
Checked with Mike Borgelt, who answered exactly this question. According to him the Cambridge 302 protocol is used, what is inline with the source (BorgeltB50.cpp includes the interface from there),  and he assumes this instrument allows only entering with fraction in 10% increments. I quickly checked the handbook of the CA302, hence did not find anything about its telling just to enter the ballast until 100%. Maybe there is some user here with a 302 and can confirm how ballast is entered in the 302, and if its like this, guess we can close this issue.  

_The B50 does not accept data to change ballast, bugs or MacCready setting. They are done from the control knobs.
That data is sent out once a second at 4800 baud.
The B500 is the same.
The B600/B800 send data at 38400 baud and receive at the same baud rate.
The input commands are exactly those for the Cambridge 302. We did this so the XCSoar developers wouldn't have to cope with yet another protocol.
I believe you can only change the ballast in 10% increments as for the 302.
The present arrangement seems to work fine when I tested it.
I hope this helps. If you need more, please let me know.

Regards

Mike_

lordfolken, 2022-08-23:
Can we close this in favor of #824 ?

iltis42, 2022-08-23:
Yes i think so. For legacy devices we won't be able to do alter this, for protocols still under development, there is an idea now how to to fix

closed 2022-08-23 by iltis42