XCSoar/XCSoar #465: Version 7.0 preview15 could not compiled for Windows

Blaubart, 2020-08-17:
I tried to compile preview15 for Windows, but I get an error:

`CCLD     curl.exe
/usr/lib/gcc/x86_64-w64-mingw32/7.3-win32/../../../../x86_64-w64-mingw32/lib/libmingw32.a(lib64_libmingw32_a-crt0_c.o): In function `main':
./mingw-w64-crt/crt/crt0_c.c:18: undefined reference to `WinMain'
collect2: error: ld returned 1 exit status
Makefile:973: recipe for target 'curl.exe' failed
make[3]: *** [curl.exe] Error 1
Makefile:2087: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
Makefile:985: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
Traceback (most recent call last):
  File "./build/thirdparty.py", line 151, in <module>
    x.build(toolchain)
  File "/home/User/Desktop/XCSoar/build/python/build/autotools.py", line 100, in build
    MakeProject.build(self, toolchain, build)
  File "/home/User/Desktop/XCSoar/build/python/build/makeproject.py", line 31, in build
    self.make(toolchain, wd, self.get_make_args(toolchain))
  File "/home/User/Desktop/XCSoar/build/python/build/makeproject.py", line 28, in make
    cwd=wd, env=toolchain.env)
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '--quiet', '-j8']' returned non-zero exit status 2.
build/thirdparty.mk:45: recipe for target 'output/WIN64/lib/x86_64-w64-mingw32/stamp' failed
make: *** [output/WIN64/lib/x86_64-w64-mingw32/stamp] Error 1`

With preview14 everything works fine.

Blaubart, 2020-08-23:
OK, I didn't know, that Ubuntu dosen't work. With Ubuntu I could compile preview14!

lordfolken, 2020-08-25:
Ok, its not an ubuntu issue. I can repoduce this with the docker container:
  CC       ../lib/curl-warnless.o
  CC       ../lib/curl-curl_ctype.o
  CC       ../lib/curl-curl_multibyte.o
  CCLD     curl.exe
```
/usr/bin/i686-w64-mingw32-ld: /usr/lib/gcc/i686-w64-mingw32/8.3-win32/../../../../i686-w64-mingw32/lib/../lib/libmingw32.a(lib32_libmingw32_a-crt0_c.o): in function `main':
./build/i686-w64-mingw32-i686-w64-mingw32-crt/./mingw-w64-crt/crt/crt0_c.c:18: undefined reference to `WinMain@16'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:974: curl.exe] Error 1
make[2]: *** [Makefile:2087: all-recursive] Error 1
make[1]: *** [Makefile:985: all-recursive] Error 1
Traceback (most recent call last):
  File "./build/thirdparty.py", line 151, in <module>
    x.build(toolchain)
  File "/opt/xcsoar/build/python/build/autotools.py", line 100, in build
    MakeProject.build(self, toolchain, build)
  File "/opt/xcsoar/build/python/build/makeproject.py", line 31, in build
    self.make(toolchain, wd, self.get_make_args(toolchain))
  File "/opt/xcsoar/build/python/build/makeproject.py", line 28, in make
    cwd=wd, env=toolchain.env)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '--quiet', '-j8']' returned non-zero exit status 2.
make: *** [build/thirdparty.mk:45: output/PC/lib/i686-w64-mingw32/stamp] Error 1
```

lordfolken, 2020-08-25:
So replacing the curl version with 7.64.1 seems to fix the compilation issue. 
```diff
diff --git a/build/python/build/libs.py b/build/python/build/libs.py
index 0117e98eca..fadd6452f9 100644
--- a/build/python/build/libs.py
+++ b/build/python/build/libs.py
@@ -109,9 +109,9 @@ freetype = FreeTypeProject(
 )
 
 curl = AutotoolsProject(
-    'http://curl.haxx.se/download/curl-7.69.1.tar.xz',
-    'https://github.com/curl/curl/releases/download/curl-7_69_1/curl-7.69.1.tar.xz',
-    '03c7d5e6697f7b7e40ada1b2256e565a555657398e6c1fcfa4cb251ccd819d4f',
+    'http://curl.haxx.se/download/curl-7.64.1.tar.xz',
+    'https://github.com/curl/curl/releases/download/curl-7_64_1/curl-7.64.1.tar.xz',
+    '9252332a7f871ce37bfa7f78bdd0a0e3924d8187cc27cb57c76c9474a7168fb3',
     'lib/libcurl.a',
     [
         '--disable-shared', '--enable-static',
```

MaxKellermann, 2021-01-22:
Fixed by a9d98bae768a5f6a764ef46804e4f9ea2718ca2d due to the parameter `-DBUILD_CURL_EXE=OFF`

closed 2021-01-22 by MaxKellermann