XCSoar/XCSoar #466: Black screen with Oppo Reno2z Android 10 Color OS V7

Adrienglide, 2020-08-21:
Hello,

It seems that XC Soar is not working with the latest version of Android (10, Color OS V7) or with the OPPO Reno 2z that I'm currently using.

When I try to start a flight or a sim, there is a few flash and then the screen goes dark.

I would like to know if you have any information about this problem?

Thank you in advance

Adrien

XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

MaxKellermann, 2020-08-21:
This cannot be solved until a developer has such a device.

groundhogxc, 2021-04-06:
@Adrienglide, can you try again with V 7.2? There was a change in the OpenGL graphics code that solved other issues. Other than that, at least a *lot* more info would be needed, but really a device like Max wrote.

Adrienglide, 2021-04-07:
Hello,

At first, thank you for keeping me in touch.

I just downloaded the new version and it works perfectly for the display,
no more black screen!

I'm not able to fly until May so I don't know if the flying data from the
GPS are consistent.

I am currently in Ireland and I am not able to download the Irish high
resolution map. The downloading start but never end and doesn't seems to be
working. Hopefully I already have the French map that I need for my flights.

Adrien Bérardan

lordfolken, 2021-06-09:
@Adrienglide the downlaod issue for android 10 has been fixed.

closed 2021-06-09 by lordfolken