XCSoar/XCSoar #477: Wrong date on flight log files

gardph, 2020-09-17:
Hi,
I'm running XCSoar on  Samsung Note 3 that's suddenly started recording the wrong dates on flight logs.  I've done a factory reset and reinstalled xcsoar from scratch but still have the problem.  The date in the trace data is also wrong reflecting that of the file name.  

Welcome! - We kindly ask that you:

  1. Fill out the issue template below - not doing so needs a good reason.
  2. Use the forum if you have a question rather than a bug or feature request.

The forum is at: https://forum.xcsoar.org

Thanks for understanding, and for contributing to the project!

-->

XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

rawtaz, 2020-09-17:
@gardph Please properly fill out the issue template instead of just adding your text at the top of it.

Also, please check your date and time on the device.

Plantain, 2020-09-17:
Probably GPS date rollover -
https://www.samsung.com/us/support/troubleshooting/TSG01200843/

lordfolken, 2020-09-17:
7 year old device. This is the gps rollover issue. Unfortunately this is only fixable with a firmware update from the manufacturer, and the manufacturer usually doesn't care.

i love how they blame the "apps" in that link, even though its them that did the error in the firmware.

closed 2020-09-17 by lordfolken