XCSoar/XCSoar #480: Display relative altitude for traffic on map

DavisChappins, 2020-09-24:
XCSoar version
--------------

6.8.16

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Traffic drawn on the map (not the FLARM radar popup or the FLARM specific radar page) shows aircraft rate of climb, if positive, below the traffic target. Relative altitude should also be shown above the target. Relative altitude should be shown in hundreds of feet (or meters) with an up or down arrow preceeding the numeric field. It can be capped at 99 if using two digits or a third can be displayed for targets greater than 9,900 ft (or meters) difference in relative altitude. Both are acceptable.

This is an EFB industry standard way of displaying traffic and relative altitude is the most important traffic attribute to be displayed.
Currently I have to press on the traffic target to see its information to get it's altitude.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

Display of relative altitude above traffic targets should be shown on the map. This information is already gathered and shown on the FLARM radar page but since that maximum range is 2km you end up with bunched up targets overlapping information when looking for relative altitudes.

Sample image here: https://i.imgur.com/V4ZFkNb.jpg taken by me driving down the road, displaying ADS-B traffic. The two red circles indicate traffic targets that I have added relative altitude to.

For people who don't wish to see relative altitude on the map, this function could be enabled/disabled by a config inside Map Display > Elements > FLARM traffic. Currently FLARM traffic is selected to **on** or **off** . It could be modified to **on**, **on with relative**, **off**.

bomilkar, 2020-09-30:
I started a discussion on displaying relative altitude:
https://forum.xcsoar.org/viewtopic.php?f=2&t=3753

DavisChappins, 2020-10-13:
Your link talks about color coding based on relative altitude. This is different.

Every EFB and TCAS display lists the relative altitude of the target above (or sometimes below) the target icon. This simply brings XCSoar to the same display standard.

lordfolken, 2021-04-08:
I'd love a pr for this.

bomilkar, 2021-04-08:
I have a patch which does this. Will submit a PR in an hour or so.
Is there a consensus on what's a reasonable relative altitude difference inside which the warning color (orange) should be used? I'm using +/- 50m and it seems OK. But sometimes I think +/- 75m would be a better choice.
BTW: does anybody know whether FLARM reports the relative altitude of the traffic based on GPS or baro? How accurate is this?