XCSoar/XCSoar #481: FLARM radar display only when traffic is within 2km

DavisChappins, 2020-09-24:
XCSoar version
--------------

6.8.16

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Currently if enabled, the 2km ranged FLARM radar overlay displays all traffic targets coming in via NMEA 0183 $PFLAA sentences, regardless of distance. For situational awareness of close targets this is an excellent display. If the traffic source is sending targets greater than 2km distance, the FLARM radar overlay still shows targets, just bunched up around the edge.

The FLARM radar page should pop up when traffic is present at less than 2km. Other targets that are greater than 2km do not need to be decluttered from the map, just the map itself should appear and disappear based on the closest target being within 2km.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

When multiple traffic targets are present the FLARM map becomes cluttered and less useable. It's fantastic for close-in awareness. There is value in seeing targets that are more than 2km away for situational awareness or tracking purposes. Looking for gliders ahead that are circling (and climbing), or watching arrivals/departures at a possible landout airport.
However, the FLARM map becomes cluttered as it populates every single NMEA 0183 $PFLAA target on the outside edge of the map. See: https://i.imgur.com/0NbtWO8.jpg or https://i.imgur.com/rWMa3PV.jpg (ignore the orange alert level 1, I set that manually)

Logic:
If any traffic in your traffic list relativeEast or relativeNorth is within 2,000m (positive or negative), then display the FLARM map. If not, hide it.

For users who don't want this functionality it can be enabled or disabled by a config at Gauges > FLARM, Other > FLARM radar.
Currently FLARM radar has **on** or **off**. It can be modified to include **on**, **on** **within 2km**, **off**.

lordfolken, 2021-03-25:
Please make it configurable with two config options. One for on off and a second one for the range. 
This is also an issue with all the ads-b traffic coming from powerflarm.

DavisChappins, 2022-08-17:
Agreed with ads-b traffic, currently the radar map is up 100% of the time with nothing displayed or one target at the edge. The radar map is great but only needs to pop up when a target is within 2km and be hidden otherwise. In XCSoar v7.x it is still up all the time regardless of traffic distance.

Unfortunately I cannot code but would like to help this issue. Is anyone working on this?

closed 2023-03-19 by lordfolken