XCSoar/XCSoar #482: Support for RASP Poland direct maps download

aggsystems, 2020-09-25:
XCSoar version
--------------

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Currently only Germany, UK and some other countries has possibility download RASP maps
-------------------------------------------------------------------

lordfolken, 2020-09-29:
In master you can add your source here:
https://github.com/XCSoar/XCSoar/blob/master/src/Weather/Rasp/Providers.cpp
That way Poland would be select-able through the list. 

In a future setup i'd like to provide it through https://github.com/xcsoar/xcsoar-data-repository

I did a sample implementation: Care to extend it?
https://github.com/lordfolken/xcsoar-data-repository/tree/rasp

dingetje, 2022-02-17:
@lordfolken did you abandon your sample implementation? I couldn't find it

lordfolken, 2022-02-17:
xcsoar-data-repository has been replaced by xcsoar-data-content. But the new strutucture should make it easier to add new datatypes.

lordfolken, 2022-08-23:
@dingetje the rasps are available in the xcsoar-data-content repository. But currently xcsoar doesn't read them. If someone would take up the work in the linked PR, it could be factored out.

dingetje, 2022-08-23:
@lordfolken is the xcsoar-data-content repository already in use for other data types in a release? The amount and tone of comments from Max in your PR do not sound promising.

lordfolken, 2022-08-23:
yes, most other things are already supplied by the repository (maps,airspaces,waypoints). (Its not my pr its @Developer66 ) but i think we could separate out the rasp feature that was written, as this is mostly extension of existing stuff.

lordfolken, 2022-09-05:
@aggsystems I have now all the things ready. The Linked PR moves all the rasp defintions to the repository. 
What i need from you is a stable link to a download dat file for your rasp site. 
Examples are here:
https://github.com/XCSoar/xcsoar-data-content/tree/master/data/remote/rasp/region

lordfolken, 2022-09-05:
https://github.com/XCSoar/xcsoar-data-content/issues/288

closed 2022-09-06 by MaxKellermann