XCSoar/XCSoar #487: A new NavBox "Event Marker"

matti5, 2020-10-15:
The new German contest rules (https://www.daec.de/fileadmin/user_upload/files/2020/Sportarten/Segelflug/Sport/Downloads/FAQ-EventMarker_Abflug.pdf) define a so-called "event maker depature". That means you have to be a event maker 10 minutes before your valid depature in th contest. After this 10 minutes of waiting (time1) you have 10 minutes (time2) to make a valid depature. The power flarm as igc logger has not a button to make a a pilot-initiated event in the igc file. But is posible to send on the serial port the sentence "$PFLAI,PILOTEVENT*" to initiated this. You get than a receipt "$PFLAI,PILOTEVENT,OK*"

Is it possible to make a new NavBox when  you push this box send the sentence $PFLAI,PILOTEVENT*7C to the power flarm an after receive "$PFLAI,PILOTEVENT,OK*" start a countdown (time1) in red up to zero a countdown start in green (time2) Also a  possibility to adjust a other time1 and time2 as 10 minutes.

glidingFisi, 2020-12-04:
I created a while ago a prototype for marker start and used it at the HWW competition.
Good Informationen that also a marker can be send to the logger.
https://github.com/XCSoar/XCSoar/compare/master...glidingFisi:feature/marker_start

lordfolken, 2021-03-13:
There is a marker function, which drops waypoints on the map. Nav-Nav-MarkDrop
It would be interesting to send the PILOTEVENT to flarm at the same time if configured.

lordfolken, 2022-04-26:
PILOTEVENT has been implemented as a Menu item. Nav(2/2). I believe this resolves this issue.

closed 2022-04-26 by lordfolken