XCSoar/XCSoar #494: Start cylinder entry

OttoPh, 2020-11-14:
XCSoar version
--------------

 Always the latest

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

 In hang gliding a start cylinder entry is often used in competitions, where Xcsoar only allows exit start cylinder to start the task. This creates problems in setting up and checking the task. But more importantly it is impossible to use the time to start countdown, so starring at the right moment is difficult. 
As all other waypoints are entries, it seems bot top difficult to add this possibility, I hope I am right about this, not familiar with programming much. 

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

 Check task (length) with given to make sure you have the right waypoints and cylinder set.
AND
Allow Start Open infobox to worm properly and start at the best possible moment

Vantuilmv, 2021-09-02:
Hello, I had a problem with this recently, I participated in the Copa Brasil de Paragliding last week and in several events the start was entry, but one day we had a waypoint with a radius of 6 km of entry (start) and within this same point the first crossing point was a radius of 3 km, and with that it didn't mark the beginning of the race, I had to advance to the next point manually, a suggestion would be to have two types of start for example: Start Cylinder (entry) and Start Cylinder (exit) .

closed 2021-09-03 by MaxKellermann

OttoPh, 2021-09-04:
I agree this is an issue.

In hanggliding it is quite common to have either an exit or entry start 
cylinder and unfortunately XCSoar has no option for this yet.

Adding this will be highly appreciated!

You may now trick yourself around it, but a countdown to the start will 
then not work for example.

Hoping on a positive response,
Otto Philipse