XCSoar/XCSoar #502: Water Ballast quantity changes in flight

ramosflys, 2020-12-07:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

enter an amount of ballast check during flight.

Do you have any idea what may have caused this?
-----------------------------------------------

Sorry no. I have used a different galaxy note 3 and have not noticed the problem. maybe a different software version is causing conflicts.

Do you have an idea how to solve the issue?
-------------------------------------------

Not Sure, I will re try the other note 3.

ramosflys, 2020-12-07:
The problem is occurring with version 6.8.17 but not happening with version 6.8.12

lordfolken, 2021-04-06:
@ramosflys can you test this with the current 72/7.3 version?

ramosflys, 2021-04-06:
Sure can. I will download and install. Will not fly for two weeks as the
weather looks bad for next weekend. Will let you know what I find out.
Thanks
Simon

MaxKellermann, 2021-08-23:
No reply

closed 2021-08-23 by MaxKellermann