XCSoar/XCSoar #503: Wave Weather forecast RASP

Hagoort, 2020-12-08:
XCSoar version
--------------

6.8

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Add Vertical Velocity data fields/overlays  to the the RASP Weather Forecast available data fields.

The Wave forecasts, in my opinion the most usefull RASP data fields while flying, are not available as a overlay in XCSoar.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
Having the wave/convergence forecasts or vertical velocity plots for different pressure levels available while flying.

Our RASP team produces free wave forecasts on a high resolution for Western Germany, see http://blipmaps.nl/NL1KM/ We produce Vertical Velocity or Wave/Convergence forecasts for different pressure levels or altitudes. We want to make these available in the cockpit while flying. We can produce the xcsoar-rasp.dat file. It is also possible to add the wave plots but they aren't shown correctly. The legend colors are not very clear. We tried many different things as the legend colors for the wblmaxmin (see figure 1) data field, also cm/s, are much better or easier to read.
[image]

We didn't succeed in changing the legend colors (see figure 2) .
[image]

After investigating further we found out that the legend colors are hard coded in https://github.com/XCSoar/XCSoar/blob/master/src/Weather/Rasp/RaspStyle.cpp One thing should be noted, to also help other RASP forecasters, the name of the datafield can change. In RASP it is possible to produce a wave/convergence plot for every pressure level. The name of the datafield always starts with press followed by the level, so press955, press899, etc. We also defined our own legend colors for the vertical velocity plots. It would be great if these can be used. I included the exact colors below. If you have any questions or if I can help in any way please contact me.
If the wave/convergence forecasts are available for different pressure levels many more great glider flights will be made. This has been proven by the comments in onlinecontest by pilots using similar datafields in commercial software.

# r       g       b
 32      32      32
 64      64      64
 96      96      96
128     128     128
 66      44     178
101      81     204
143     126     229
191     178     255
 15     107     153
 44     133     178
 81     163     204
126     195     229
178     229     255
255     216     178
229     177     126
204     142      81
178     111      44
153      84      15
255     178     178
229     126     126
204      81      81
178      44      44
253     157      24
253     189      16
253     224       8
253     253       0

Hagoort, 2020-12-09:
I would say my coding skills aren't that much. But do I understand correctly that the only thing that has to change to get the same legend as the wblmaxmin datafield is the following. In https://github.com/XCSoar/XCSoar/blob/master/src/Weather/Rasp/RaspStyle.cpp the term "nullptr" in line 138 defines what to do with "new" datafields that are not hard-coded or unknow? If so than by changing "{ nullptr, rasp_colors[0], 2, false }" into "{ nullptr, , rasp_colors[5], 1, // max range 256*(1**2) = 512 cm/s = 5.0 m/s false }" would produce the same legend as wblmaxmin for the wanted pressXXX datafield maps. This would help. I will dig deeper into this to propose changes to get the legend exactly as wanted or similar as the one used on the blipmaps forecast website. I would be nice if someone could check and react if I'm right.

roscosoarer, 2020-12-10:
The colour schemes produced in SkySite are much more pronounced and may suit your needs.  In Australia these are also available on GFAMet.  I was working on creating a .xcm file, load that into 'XCSoar Sitefiles' as a terrain file in the hope that the forecast wave profiles would become terrrain.   I have no idea how to create .xcm files from the RASP weather data.

Just  a thought, from outside yor box.

[image]

Hagoort, 2020-12-11:
Thanks. Did you check the colour scheme we use for wave/convergence? Also suited for red-green colorblind people. If GFAMet is based on RASP than it is easy, for the people running the model, to get the forecast maps in XCSoar. Check the XCSoar manual. No need to create .xcm files.

Hagoort, 2020-12-11:
There are a few solutions. If "nullptr" in line 138 defines what to do with "new" datafields that are not hard-coded or unknow, I will add another color scheme and have nullptr use it. As with RASP it is possible to produce a plot for every pressure level there is no need to hard-code any or all level and it will always use the newly defined scheme. Also I don't know how to hard-code pressXXX into somekind of variable that works for every pressure level.

Hagoort, 2020-12-11:
This is how our color scheme looks:
[image]

lordfolken, 2021-03-25:
> I would say my coding skills aren't that much. But do I understand correctly that the only thing that has to change to get the same legend as the wblmaxmin datafield is the following. In [...]

the nullptr is a catch all. So if the field name doesn't match it will use rasp_colors[0]. The digit in rasp_colors simply refers to the position of the color ramp in the constant expression ColorRamp. 

You can add your defintion to the end of the const expr ColorRamp and refer to it with a field name.

dingetje, 2022-02-17:
I will have a go at this one, would it be acceptable to have an external loadable color map file that can be assigned to the parameter of choice? This in addition to the already existing hard coded color ramps.

In general the RASP overlays are way to "hard coded" and require some sort of configuration. Maybe this can be resolved if ticket #768  is implemented.

dingetje, 2022-09-06:
@lordfolken I've noticed the RASP data sources are now no longer hard coded (good), but the RASP styles still are (hard coded). What is your take on this?

It doesn't seem too hard to implement loading an optional color legend file as requested by @Hagoort

lordfolken, 2022-09-06:
If then it would have to be a style file that is supplied inside the xcsoar-rasp.dat archive. That way whoever wants to extend the layers or color schemes can do so. If the file isn't found the harcoded fallbacks are used. So the current format would be untouched. 

The caveat of that is that everyone will do their funky color scheme. 

The manifest describing the colors / zoom levels for the topology layers inside the XCM map archives has a similar function: https://mapgen-data.sigkill.ch/manifest
This can also be extended (additional layers) and modified (colors, zoom levels etc) and xcsoar parses it.

The TopographyStore::Load Function seems to be responsible for parsing that file.

I just completely lack the time to implement this. If someone would start at least i can occasionally lend a hand.

lordfolken, 2022-09-14:
I reworked @Developer66 PR for thermal map's rasp files. 
https://github.com/XCSoar/XCSoar/pull/907/commits/ed6c742a406d3c991da74e2e136a4bfd37d6a737

This includes now many new layers including:
https://github.com/XCSoar/XCSoar/blob/ed6c742a406d3c991da74e2e136a4bfd37d6a737/src/Weather/Rasp/RaspStyle.cpp#L462-L491
I have also used names for each, instead of array index numbers. This makes it more readable, and more easy to understand for non programmers. 

Once (if) this is merged, many more styles and layer options should be available for usage by RASP forecasts.

While this isn't a custom style definition, i think your files could be adapted to fit thermalmap's style. 

Can you work with this @Hagoort @dingetje ?

PS: @Hagoort can you provide a link to blipmaps.nl's rasp file? so we can add it to https://github.com/XCSoar/xcsoar-data-content/tree/master/data/remote/rasp/region?

closed 2022-09-15 by MaxKellermann