XCSoar/XCSoar #505: Upload IGC to WeGlide

bomilkar, 2020-12-28:
XCSoar version
--------------

7.0

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

We will see a new competition platform in the coming season, at least in Germany: DMSt is committed to switch to WeGlide.
I learned from the WeGlide team that there is an API for uploading IGC files.
https://api.weglide.org/docs#/igcfile/upload_flight_simple_v1_igcfile_post

**I'm proposing to integrate an Upload feature into XCSoar.**

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

I'm envisioning adding the choice to upload the flight as part of the "Flight download" button/function in the logger dialog.
For example with FLARM as a connected logger:

- hit the "Flight download" button
- select a flight
- the download of the IGC file happens
- after the download  is complteted and **_before_** you see the message "Do you want to download another flight?"
- **insert the choice to upload the IGC file to WeGlide.**

It may need an extra page in Setup to define WeGlide Profile data. Alternatively it may collect/remember this information (Pilot ID, account info, ...) as part of the choice to upload to WeGlide.

Has anybody started to work on this already?
Maybe build this feature such that it can deal with other platforms, too. For instance SkyLines.

moldhouse, 2020-12-28:
Hey, just dropping a note from WeGlide. We have no familiarity with the XCSoar source code and can not state if this feature is sensible in any way.

If this feature is wanted for the XCSoar project, we are happy to provide help or adjust API endpoints. LXNAV and SeeYou also use our API to fetch tasks (cup or json) which users have planned.

MaxKellermann, 2020-12-28:
What is WeGlide? It has a `.org` domain, but it isn't an "organization". It is a commercial company, funded by tax money, it appears. If it was funded by the taxes I pay, I'd expect it to be a charitable/non-profit, but it is not. The code is proprietary and closed, isn't it?
This is unlike the OLC - which is a "gGmbH", which is a *non*-commercial company (officially). (Also proprietary and closed-source, but weird in another way - many people work for the company for free, while the company collects massive funds from the gliding association, effectively paid by the same people contributing their time for free.)

bomilkar, 2020-12-29:
I understand what you mean, @MaxKellermann . And you raising a valid point that I hadn't thought of. But I don't want to speculate about how WeGlide is funded or if anyone is making money off of it. I want to hear that first hand.
Anyway, this interface is a tiny addition that won't make a big impact on either side. The intent is to improve the user experience. It won't make pilots use WeGlide if they haven't already opted in.
And by the way, XCSoar has a lot of interfaces with commercial products and devices. One more or less will not make a difference. I would agree with you if a for-profit organization were to bag/borrow/steal XCSoar to make their Thing a product. But that doesn't seem to be the case.

moldhouse, 2020-12-29:
We receive funding for our ideas regarding research on flight data. We would not have gotten the funding without setting up a company and we hope that this will enable us to work full-time on the project.
I understand that this might not be the path that you would have taken but I rather go down this path than stick with the systems we have forever.
We will open source many of our projects in the coming weeks and months. As I have stated in my comment above, I understand that there are reasons not to integrate for an open source project.

August2111, 2021-01-02:
Hi Bomilkar, I can very understand your point of view on this problem - and I fully share it! As far as I understand it, the guys from WeGlide from the DAeC have already been awarded the contract for the DMSt evaluation in 2021. I cannot say whether that's good or bad, but I think it should be when the DMSt championship starts on March 1st XCSoar also use the opportunity to offer a simple upload interface!
That's why I would like to program a corresponding feature according to your suggestion as soon as possible - and that's why I have already contacted WeGlide. I will deal with it further in the coming days / weeks and hope to be able to offer a first solution around mid-February.
BTW .: Unfortunately, a similar solution was not offered or not wanted by the OLC!

lordfolken, 2021-03-25:
Btw.. OLC seems to have reversed its stance this year and now allows uploads.

bomilkar, 2021-03-26:
OLC: Too little, too late.

lordfolken, 2021-03-26:
> OLC: Too little, too late.
i wholeheartedly agree there. I'm confident weglide will eclipse olc within 2 years.

VladimirVLF, 2021-04-19:
We will see who is who when WeGlide funding will be fully used.
OLC is a dinosaur, yeah.

lordfolken, 2022-11-15:
I believe this to be implemented now. Therefore closing this issue.

closed 2022-11-15 by lordfolken