XCSoar/XCSoar #508: TE compensated final glide calculation

christoph3761, 2020-12-30:
XCSoar version
--------------

7.0

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

TAS data should be used in order to compensate for speed changes in final glide calculations (total energy compensation). This is actually described in section 6.9 of the manual. But referring to https://forum.xcsoar.org/viewtopic.php?f=2&t=2510 and the XCSoar changelog, it has never been implemented.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

In current versions of XCSoar, the final glide calculation varies quite heavily with changes in airspeed. So if you pull up, your final glide increases, if you speed up, it decreases. This happens even though TAS data is available through external devices and it leads to inprecise final glides. I wish the final glide calculation of XCSoar would be more accurate. A TE compensated final glide calculation is more accurate since it is mainly affected by change in total energy, wind or glider performance. I think this feature is of significant value for competition gliding.

groundhogxc, 2026-04-14:
Great, I have wanted that for some time. Just one comment on the implementation in #2398 (which actually has more in it than just this, so I am commenting here) - if I understand correctly, right now the energy height is used (added) directly to calculate the correction. This however implies that the target point is reached at velocity 0 - not a legal flight state to be in unless you have an aerobatic license :). Or, in other words, just adding the height leads to a too low actual arrival by about 50 meters in typical cases.

This is what I think I found though - maybe I overlooked something, and it is actually taken into account, in which case ignore this post.

The algorithm should assume a reasonable final speed on arrival - and that is what will make about 50m difference in the calculation, for most situations (approx 100 km/h speed).

So basically, the energy height for the current TAS should be added to the altitude, and the energy height corresponding to a fixed reasonable velocity should be added to the height of the final point for the calculation (or subtracted from the initial altitude), to compute a more useful AltD or Arrival Alt etc.

The "reasonable velocity" should probably be the McReady speed for the current MC setting. Or maybe the safety MC? Something like that, I guess?

lordfolken, 2026-10-18:
I'm moving this to v7.47. There was strange behavior on a test flight, and i therfore don't want to influence something so central, so close to a release.