XCSoar/XCSoar #510: Issues with graphical output on Raspberry Pi 4 (V7Prev15, 64bit arch)

HerbyDE, 2021-01-01:
XCSoar versions having and not having the problem
-------------------------------------------------
XCSoar Version 7 Prev. 15 on Raspberry Pi 4 (4GB, Rev. 1.1)

Expected behavior
-----------------
XCSoar is launched when invoking "xcsoar".

Actual behavior
---------------
I compiled the source code as given in the developer manual without any extra flags or variations. 
Upon invoking "xcsoar" this error occurs:
`Free storage 0 KB
Starting XCSoar XCSoar v7.0_preview15-Linux~git#262bd4b2bf
Using DRI device /dev/dri/card1 (use environment variable DRI_DEVICE to override)
Display dpi=55,58
OpenGL: mda=1 npot=1 fbo=0 stencil=0
Initialise fonts
drmModePageFlip() failed: -13`

Also, it says "PageFlip", which to my understanding is a sign for a rotated screen. I did not rotate the screen yet, since I have read about issues with rotation before.

Steps to reproduce the behavior
-------------------------------
XCSoar got compiled on the RPi 4 itself without any flags. I use the latest 64bit Raspbian as OS.
It does not matter which HDMI output is used. The error remains the same.
No extra connections, software, etc. involved.

Do you have any idea what may have caused this?
-----------------------------------------------
The Pi offers two graphics output devices to my understanding: card0 and card1. Even though the error message says overriding is possible with an ENV VAR, it still takes card1.

Do you have an idea how to solve the issue?
-------------------------------------------
I guess, as stated above, the addressed video output causes the error.

bomilkar, 2021-03-14:
The issues have been described here:
https://forum.xcsoar.org/viewtopic.php?f=30&t=3741
along with some solutions and workarounds.

closed 2021-03-14 by MaxKellermann

saulolvn, 2021-04-16:
-Hi guys, I would like to ask you guys for this solution, since I am trying to fix an almost the same problem on raspberry pi, but I don't know why I just can't get into the forum for the solution, could you guys kindly post it here?
-My issue informations:
After a bit hard time installing xcsoar on raspberry pi 4 (pi4 8gb version) I get his message here:
Using DRI device /dev/dri/card1 (use environment variable DRI_DEVICE to override)
drmModePageFlip() failed: -13

Thank you.

bomilkar, 2021-04-16:
If it's not /dev/dri/card1 it might be /dev/dri/card0
So in your case insert a line into .profile before you start XCSoar
`export DRI_DEVICE=/dev/dri/card0`

If it's none of these 2 devices you may not have DRM VC4 V3D driver enabled.
Make sure this line is in /boot/config.txt
`dtoverlay=vc4-fkms-v3d`

You say: "can't get into the forum for the solution". What exactly is the problem?

saulolvn, 2021-04-17:
Thank you bomilkar for your response.
-The .profile solution didn't worked, that drove me to dig more about it and found the developer's manual, there I found out that there are a lot of libraries packages that should be installed to developers to work with, so for my case what I did was that every Linux and/or Raspberry libraries it asked to install for the Raspberry Pi4 the manual says to use I installed. 
(I did changed the dtoverlay=vc4-fkms-v3d as you and the normal manual said to do)
After that xcsoar 7.4 arm64 version is now working on my setup. Thank you so much, I'm new to this gliding world, really loving it. So I would like to help this community! Love programming for micro PC's like Raspberry PI, I'm still new to this community too. 
After this experience I would like to suggest that the normal manual at the raspberry pi installation section has a small sub section like: Every library you need to install before installing XCSoar package.

-About the XCSoar forum bomilkar every time I try to enter the browser returns to me this message:
Forbidden
You don't have permission to access this resource.
Apache/2.4.38 (Debian) Server at forum.xcsoar.org Port 443
A friend of mine get this message too, I don't know why this is happening, we were discussing if there is any ip blocking or something like that, we are from Brazil.

-Just a note: my goal is to integrate a HAT board called Navio2 (for UAV's sensoring and that comes with Ardupilot from the company EMLID https://navio2.emlid.com/) and an airspeed sensor for great real time data for gliders.
Once more Thank you very much for your help.
Here willing to help spread this great platform and contribute to this commuity.