XCSoar/XCSoar #511: Can't select airfield / waypoint on high display resolutions

denis-robel, 2021-01-02:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

bomilkar, 2021-02-03:
Your device may think you have a pointing device like a mouse.
Instead it should think it has a touch screen. This widens the click radius.
You try this by setting n_touch_screens = 1 in src/ui/event/poll/libinput/LibInputHandler.hpp

GLDRLS, 2021-03-04:
hello bomilkar
Thank you for your answer.
I ran into the same problem
set n_touch_screens = 1 in src / ui / event / poll / libinput / LibInputHandler.hpp
unfortunately did not help

is there another option that can be tried?

bomilkar, 2021-03-04:
@GLDRLS : this does the trick for OpenVario. I can't help if it doesn't do it for you.

GLDRLS, 2021-03-04:
Thanks for your quick reply

I tried a few things.
And for me a udev rule works
I did extend my udev rule for rotating the touchscreen orientation

Extended with the following line
ENV {ID_INPUT_MOUSE} = "1"

I got it from her :
https://people.freedesktop.org/~whot/libinput-rtd/device-configuration-via-udev.html

MaxKellermann, 2021-03-15:
> the size of area to select an airfield should depend on scaling by screen resolution (dpi) and symbol scaling for airfields

This is already what happens! XCSoar uses a physical hit radius, which is larger for touch screens.

bomilkar, 2021-03-15:
I agree, @MaxKellermann . But obviously it doesn't address the issues users are facing.

I can think of 4 different pointing devices being used with XCSoar:
Mouse, touch screen, 4-way switch, rotary encoders.
They differ in 2 characteristics:
1. how accurate one can move to a position
2. how long does it take to click the select button once the pointer is in position

A mouse is accurate and fast.
Rotary encoders are accurate but slow.
Touch screen is fuzzy and reasonably quick.
4-way switch is not as accurate as a mouse, and not as fast as a mouse.

The solution to the (lack of) accuracy is a wider click radius.
The solution to longer click time is not so obvious. I had started some experiments with "relative move". For example: when your pointer is above the map the positioning happens relative to the moving map instead of relative to the display frame. That means you can position the pointer to a point on the map (say an airport). As long as you are not manually moving the pointer it will move together with the map. Now you have time to click the select button. The "relative move" might be active for some time after the pointing device has been moved (say 10 sec). When that time expires it falls back to the traditional behavior. I hope this makes the concept of "relative move" sufficiently clear. I will throw this into the forum to see what users think.

However, I have no idea how to discover which of the 4 (or so) different pointing devices the system has. I'm reluctant to propose yet another parameter in the settings. But what else can we do?