XCSoar/XCSoar #512: UX - Flarm radar - disabled button - tapping twice opens menu

r0w4n, 2021-01-03:
Which XCSoar version are you using when seeing the problem? 6.8.17
Which XCSoar version is the last one you know that did not show the problem? I'm not sure. Probably all 6.8

Expected behavior

Given that I'm on the flarm radar screen 
And that I have zoomed out to the maximum level 
And that the zoom out button now disabled 
When I keep tapping on the zoom out button 
Then nothing happens 
And the menu is not displayed 

Actual behavior

Given that I'm on the flarm radar screen 
And that I have zoomed out to the maximum level 
And that the zoom out button now disabled 
When I keep tapping on the zoom out button 
Then the menu is displayed 

Steps to reproduce the behavior
As above 

Do you have any idea what may have caused this?
The disabled button doesn't accept the tap/click and do nothing. Instead the menu activation mechanism receives the input 

Do you have an idea how to solve the issue?
Allow the disabled button to receive the tap/click but do nothing

MaxKellermann, 2021-03-02:
Why?

r0w4n, 2021-03-03:
The situation is when you want to fully zoom in or out. You don't know how
many taps to go so you keep tapping until the button becomes disabled. This
indicates that you've zoomed fully in or out.

Inevitably when tapping this way you tap one too many times. And so it also
brings up the menu buttons because tapping on the screen twice actions
this. This is an undesirable behaviour.

lordfolken, 2021-03-25:
I see your point. Mainly a usability issue.
Its also the case for the buttons in the menu itself. For example VEGA or FLARM Radar.
Please investigate and submit a PR.

closed 2026-10-10 by lordfolken