XCSoar/XCSoar #513: Screenshots for F-Droid

mondlicht-und-sterne, 2021-01-04:
You can add Screenshots for F-Droid, too, if you want.

https://f-droid.org/en/docs/All_About_Descriptions_Graphics_and_Screenshots/

https://gitlab.com/snippets/1895688

lordfolken, 2021-02-05:
@MaxKellermann We should switch to Fastlane appstore deployment. That way we would have the stuff here in the repo, and could contribute. Would you support an initative in that direction?

MaxKellermann, 2021-02-05:
I have no idea what that means.

lordfolken, 2021-03-25:
@mondlicht-und-sterne please provide some and submit a pr.

closed 2022-08-22 by mondlicht-und-sterne

Korb, 2023-03-04:
Screenshots missing:

[image]

https://f-droid.org/packages/org.xcsoar/