XCSoar/XCSoar #515: Touchscreen inactive on Kobo Glo HD

druide862, 2021-01-17:
XCSoar versions having and not having the problem
-------------------------------------------------
Versions having the problem:
- 6.8.17
- 6.8.16
- 6.8.15
- 6.8.11

Expected behavior
-----------------
While in "Fly" mode, when I press the big **M** button, I expect the menu to show.

Actual behavior
---------------
When I start the Kobo, I first have the splash screen with FLY/SIM. I hit the FLY button, and at this time, the touchscreen is responding as expected. Then XCSoar loads lots of things (Maps, terrain, etc...) and eventualy displays the map with the gages and the big **M** button. Then, when I hit the big M button, nothing appends. Can't neither zoom in or out the map. Looks like the touchscreen is inactive (but it worked at the FLY/SIM splash screen). The kobo and xcsoar are working correctly otherwise: gages values are evolving, the map evolves as I move, etc.

So, I then power off the kobo by holding the power button 10s., and I start it again. Hit the FLY button, and then the touchscreen works as expected: **M** button shows the menu, etc...

If I power off/on the kobo within few minutes after the "hard" power off, the touchscreen works as expected **BUT**, if I let the kobo powered off a few hours, and power it on again, the problem comes back: Touchscreen inactive until I "hard" power it off.

Tested on 4 different kobo Glo HD with blue fly vario module.

Steps to reproduce the behavior
-------------------------------
- Install xcsoar 6.8.17 on a kiobo Glo HD with bluefly vario module
- power it off and let it powered off a few hours
- power it on
- hit the FLY button
- There you are: Touchscreen inactive
- Hold power button for 10s to hard power off the kobo
- Power it on again
- Hit the fly button
- Touchscreen now active (till next long powered off period...)

Do you have any idea what may have caused this?
-----------------------------------------------
No.

Do you have an idea how to solve the issue?
-------------------------------------------
Neither

lordfolken, 2022-02-24:
@GeekGoneOld this is now fixed?

GeekGoneOld, 2022-02-25:
> @GeekGoneOld this is now fixed?

As I mention in the commit, it may be fixed but I don't have a Glo HD so I can't be sure.  The particular problem mentioned in this seems to be somewhat different from any other reports so I doubt this fixes it.  It is quite possible that the problem is specific to the author of this issue and not experienced by anyone else.

@druide862 can you try the latest version of XCSoar and see if this issue is fixed?  If not, let us know if the symptoms have changed.

lordfolken, 2022-05-30:
@druide862 @GeekGoneOld as there is no reaction, i'm closing this bug report. In general i would like to have a kobo tester group, i have sent you invites, so that releases can be tested on kobos.

closed 2022-05-30 by lordfolken