XCSoar/XCSoar #521: Support tiles server and cache tiles format (such as MBTiles)

scls19fr, 2021-02-01:
Hello,

It will be a great feature to have if XCSoar could support tiles server (WMS, WMTS) and, for offline usage, support cache tiles format such as [MBTiles](https://wiki.openstreetmap.org/wiki/MBTiles) (this kind of file can be generated using [MOBAC](https://mobac.sourceforge.io/).

Any opinion about such a feature?

Kind regards

lordfolken, 2021-03-25:
Its lots of work. XCSoar's map render-er is cross-platform written in c++. So it cannot use the map render-er for example used by android. So no easy re usage of libraries.
Also a unique feature is shading of hills according to wind or sun, that cannot be done with tiles.

scls19fr, 2021-03-26:
> Also a unique feature is shading of hills according to wind or sun, that cannot be done with tiles.

Having a feature to change displayed map depending of user preference could be a solution (ie user could use either the current XCSoar vector map with the shading of hills or tiles map without such shading).

About the cross-platform aspect of XCSoar and given it's written in C++, I can understand that you can't rely on Android map render, but I can suggest using a C++ library such as [QtPBFImagePlugin ](https://github.com/tumic0/QtPBFImagePlugin) it's a [Qt](https://www.qt.io/) library which is used by [GPXSee](https://github.com/tumic0/GPXSee)

I can also understand that you don't want to rely on Qt.

This library can (at least) be used as an inpiration (it's LGPL-3.0 licensed so it's compatible with XCSoar  GPL-2.0 license).

scls19fr, 2021-04-08:
in France, Skydreamsoft provides SDOACI https://sdoaci.skydreamsoft.fr/ a VFR map (WMS)
There is also IGN Geoportail OACI VFR https://www.geoportail.gouv.fr/donnees/carte-oaci-vfr (WMTS)

lordfolken, 2026-09-29:
Triage: this is a real feature (TMS/WMTS/MBTiles under the map), but it is a large piece of work. XCSoar’s map is its own renderer, so this is not a matter of dropping in an Android or web tile view, and hill shading would not come along with the tiles.

Moving it to `backlog` rather than leaving it in needs-triage. Not closing — still wanted, just not a near-term item unless someone picks it up with a design that keeps the existing terrain shading.