XCSoar/XCSoar #525: Wrong GPS-altitude when running on Lenovo Tab M8

c63-hartmannc, 2021-02-08:
XCSoar versions having and not having the problem
-------------------------------------------------

6.8.17-Android (Latest version from Play Store shows the problem)

Expected behavior
-----------------
XCSoar should report correct GPS-altitude

Actual behavior
---------------
When opening XCSoar the GPS-altitude is reported as 485m MSL at my home, all other GPS-test-tools running on same hardware are showing 533m MSL. This later value is correct, since the height MSL is 530m (and is also reported from Infobox within XCSoar). GPS-test-tools report about 16 satellites in use, accuracy of about 4m (GPS data by examobile / GPS Test by Chartcross)

Based on wrong height, also height above ground is reported as "-45"

Steps to reproduce the behavior
-------------------------------
- Install XCSoar on Lenovo Tab M8 (Typ TB870SF, Android 10, Kernel 4.9.190+ / 15.10.29)
- Start in "FLY"-mode => altitude is reported wrong

Do you have any idea what may have caused this?
-----------------------------------------------
No, but running XCSoar (same version) on my mobile (Samsung S7 Edge) all works fine and the values reported are correct. Maybe problem related to some missing hardware-features (Tablet has no barometric or magnetical sensor)

Do you have an idea how to solve the issue?
-------------------------------------------
In case of some debugging is required (e.g. using special XCSoar-version containing attional messages or by setting up some options) please let me know. Attached the current logfile (just opening XCSoar, wait for GPS-fix, validate wrong altitude end exit)

[xcsoar.log](https://github.com/XCSoar/XCSoar/files/5945708/xcsoar.log)

groundhogxc, 2021-03-03:
This might be a red herring, but these numbers look suspiciusly like a misapplied geoid correction on the incoming GPS data. 

Since you use meters and not feet you are probably in Europe, where -45m would be a typical error that you see if you accidentally apply the geoid correction twice (e.g. the GPS delivers already corrected data, but another correction is applied later, for example by XCSoars geoid code). 
If you are in the US, then it is the other way round, it would mean that the correction is not done, because the geoid offset in north America is a similar value, but negative.

However, I do not know though how this would be cause; I thought that on Android, XCsoar polls the Android location manager for GPS values (which I assume takes care of the correction), and does not see the NMEA stream from the GPS module. Therefore I would have thought there is never a geoid correction applied inside XCSoar when using Android internal GPS. So, take with a grain of salt.

lordfolken, 2021-03-25:
can you use GPS Test or similar to determine whether its the firmware of the GPS?

em-hartmannc, 2021-04-04:
I checked with GPS Test (as given in the initial description) and all other GPS-tools report correct height, so the assumption of user groundhogxc might be correct, it looks like XCSoar performs some correction even the GPS-data is correct (and yes, my location is in Europe, close to Munich).

lordfolken, 2021-04-06:
@em-hartmannc is this still an issue with 7.2?

em-hartmannc, 2021-04-10:
Checked with XCSoar 7.3 and XCSoar-Testing 7.2, both versions still have that problem

SP-AWN, 2021-04-22:
Same issue here. Device Xiaomi MI 8 pro. GPS height is of by -30m 
Area: Poland
XCsoar version  7.4 
XCsoar GPS height 177m
XCsoar H AGL -30m

GPS Test 
height with "Apply geoid value at the current location to altitude" option unchecked: 216m (correct)
height with "Apply geoid value at the current location to altitude" option checked: 176m (incorrect)
Best Regards
Adam Wnek

teleboy2410, 2021-05-03:
Hello, 

I have the same problem with the lenovo tab m8. The gps system sends the correct altidute and xcsoar adds the WGS84 correction once more. Is it possible to calibrate the altidute or to disable wgs84 correction in xcsoar? 

Best regards, Alexander

lordfolken, 2021-05-03:
perhaps   https://github.com/XCSoar/XCSoar/blob/464df291983948c67e967725e613b25c3efd9814/src/Android/InternalSensors.cpp#L266 needs to be improved.

lordfolken, 2021-05-03:
https://developer.android.com/reference/android/location/Location#setAltitude(double)

teleboy2410, 2021-05-16:
I deleted the following sentence: auto GeoidSeparation = EGM96::LookupSeparation(basic.location); and in the following sentence (basic.gps_altitude = altitude - GeoidSeparation;) I deleted the "- GeoidSeparation; 

After that I compiled to Apk and installed the App new on my device! 

Now the altidute on my Lenovo Tab M8 is correct! Thank you for your help.

MaxKellermann, 2021-08-16:
So the Lenovo Tab M8 applies Geoid separation before giving apps access to the location data.
However, see https://developer.android.com/reference/android/location/Location#getAltitude()
> Get the altitude if available, in meters above the WGS 84 reference ellipsoid. 

This means that the Lenovo Tab M8 is buggy & broken. It must not apply Geoid separation. It is correct that XCSoar does it "again", because it should not have been done already in the first place.

XCSoar doesn't support known-broken hardware. There will not be a workaround, because that's the road to maintenance hell. Sorry.

closed 2021-08-16 by MaxKellermann