XCSoar/XCSoar #527: Kobo CP2104 baud rate

nbonniere, 2021-02-17:
XCSoar versions having and not having the problem
-------------------------------------------------
Kobo-mini with version V6.8.11-kobo

Expected behavior
-----------------

With Kobo in OTG mode and connection to USB-serial device, it works OK with FTDI FT232 device, but NOT with CP2104 device.

Actual behavior
---------------

When connected to a TTGO T-Beam which uses a CP2104 USB-serial device, the Kobo does connect OK, but the 'monitor' window just shows dots '........' as if the baud rate is wrong. I know for sure the baud rate is 38400 as I tried connecting to a PC at 38400 and it displays NMEA data.  Looks like the CP2104 does not get programmed with the right baud rate. I tried various other baud rates, but none work.

Steps to reproduce the behavior
-------------------------------

Set Kobo for OTG and connect to CP2104 device.

With same Kobo-mini in OTG and connection to FTDI FT232, it works perfectly.

Do you have any idea what may have caused this?
-----------------------------------------------
No. Probably unsupported device in Kobo drivers.

Do you have an idea how to solve the issue?
-------------------------------------------

CP210x driver seem supported, but perhaps needs a specific CP2104 driver for Kobo ?   Fix command to set baud rate for CP2104 ?

nbonniere, 2021-02-17:
Tried latest release v6.8.16-kobo. Same issue.

lyusupov, 2021-02-21:
The XCSoar for Kobo-mini is operating on top of Linux Kernel 2.6.35.3

```
~ # cat /proc/cpuinfo
Processor       : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 159.90
Features        : swp half thumb fastmult vfp edsp neon vfpv3
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x2
CPU part        : 0xc08
CPU revision    : 5

Hardware        : Freescale MX50 Reference Design Platform
Revision        : 50011
Serial          : 0000000000000000
~ # uname -a
Linux (none) 2.6.35.3-00004-g0f1bf05 #8 PREEMPT Tue Feb 11 00:01:54 CET 2014 armv7l GNU/Linux
~ #
```

There is no support for Silicon Labs CP210**4** in that old kernel:
https://github.com/embeddedarm/linux-2.6.35.3-imx28/blob/master/drivers/usb/serial/cp210x.c

CP210**2** was the only supported at that time.

MaxKellermann, 2021-03-02:
I don't think anybody will ever hack that ancient Kobo kernel again.

nbonniere, 2021-03-06:
Yes, that's unfortunate. I thought perhaps a script could be used to send a command to USB device by VID,PID or bus number. Looks like instead of setting dividers for baud rate of CP2104, the baud rate itself is used (?) Confusing datasheet/documentation...

lordfolken, 2021-04-06:
@nbonniere can you decide what you want to do with this issue? I also don't see the benefit of investing time here.

nbonniere, 2021-04-06:
I initially thought a new driver could be included, but re-compiling the Kernel is a bit much. It means that the SoftRF TTGO T_beam module cannot be connected by USB to XcSoar.

CazYokoyama, 2021-04-06:
Linar is correct. Kobo-mini's Kernel, 2.6.35.3 supports CP2101/CP2102/CP2103, not CP2104 unfortunately. There are many updates on Its driver file, cp210x.c between 2.6.35.3 and the one which supports 2104, but the one which supports 2104 seems  build-able, i.e. back porting.
The point is how the benefit of investing time on that as lordfolken says because (XCSoar + Kobo Mini + SoftRF mini) is working, http://www.caztech.com/glider/softrf/mini/KoboMini/KoboMini-ab02s.mp4

closed 2022-07-20 by lordfolken