XCSoar/XCSoar #530: Allow waypoints over water

Go77, 2021-02-23:
XCSoar version
--------------

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

lordfolken, 2021-03-13:
Waypoints are an independent layer of the map. You can set any coordinate you'd like.

Go77, 2021-03-13:
It seems that the map and the displayed waypoints are not independent. I generate maps for XCsoar via the XCSoar mapgenerator and use the high resolution format. I do not include waypoints during map generation. Below I show two screenshots. The first one is including the map, for the second screenshot I left the xcm-file out in the map-settings, so there is no map. The area you see is part of the IJsselmeer in the Netherlands. First of all you can see that the map is in error. The part of the IJsselmeer North of the dyke between Enkhuizen en Lelystad is shown as land. (This error is adressed in another bug-item (#531)). In order to highlight the problem I generated a number of waypoints with varying latitude. One group is seen just east of 'Morra'. The Northernmost waypoints are over land, the Southernmost waypoints are positioned into were the IJsselmeer is. All generated points are shown here, not a single one disappears. Apparently XCSoar concludes in some way that this is 'land'. (I think maybe there is an error in some water/land file, as the map-color also is incorrect). I generated the same sequence of 6 waypoints South of Enkhuizen. What you see now is only 2 of the 6 waypoints (one just visible in the lower left edge of the letter 'E'). The waypoints over water are not shown. (In fact the map shows the 2nd waypoint over water, but I think this is just at the edge of what XCSoar still discriminates to be water.)

Now in the second screenshot I left the mapfile out. Something strange happens to the waypoints: some are still shown, others are no longer visible. 

I hope somebody has a solution for this strange behaviour or has a suggestion how to show all generated waypoints and how to represent the correct water areas.

[image]
[image]

lordfolken, 2021-03-13:
So i think i understand what is happening:
 - XCM Map files are just zip files without compression. 
 - If you click "Include Xcsoar waypoint database" in mapgen, it will include the waypoints from what was welt2000. 
 - Inside the xcm file you find a waypoints.cup
 - If you deconfigure the map file, that waypoint file is also no longer displayed. 
 
If you load the map file and do display-display->topo off and terrain off, you should still see those waypoints.

Go77, 2021-03-13:
Dear lordfolken,

I did not include waypoints during map generation. Instead, I have my waypoints defined in two separate waypoint files. I tried switching topo and terrain but this makes no difference in displayed waypoints.

Go77, 2021-03-14:
I can imagine that polygons are used inside Xcsoar to indicate large water areas. But I would like to be able to switch this behavior off, such that I can define waypoint wherever I want. Or do these polygons also define where the map colors blue?

lordfolken, 2021-03-15:
XCsoar doesn't have a concept of water. Its just a shape (polygon) that is colored blue. It doesn't get treated any different than for example the yellow polygon that indicates a city. 

You can fly over an area of water and then add a waypoint via the waypoint editor. With the new function you have the waypoint at your current position.

Go77, 2021-03-16:
Hmm. Apparently this is not just a matter of colouring. Not only do the waypoints disappear on the map (which might occur if the colouring overrides the waypoint layer), but also the waypoints over water disappear in the waypoint-list. So, it seems they are really designated 'invalid' and removed by XCSoar.

MaxKellermann, 2021-09-24:
No reply. Will reopen as soon as there is enough data.

closed 2021-09-24 by MaxKellermann