XCSoar/XCSoar #531: Sea represented as land

Go77, 2021-02-23:
XCSoar versions having and not having the problem
-------------------------------------------------

6.8.17

Expected behavior
-----------------

Sea area is represented as land

Actual behavior
---------------

A part of the IJsselmeer in the Netherlands (between the Afsluitdijk and the Houtribdijk) is falsely represented as land area instead of water.

Steps to reproduce the behavior
-------------------------------

Pan to the area and you will sea ;-) how it looks like.

Do you have any idea what may have caused this?
-----------------------------------------------

-

Do you have an idea how to solve the issue?
-------------------------------------------
-

MikeBravo15, 2021-02-24:
Looks like a map issue. Maps are generally generated automatically with a script. Unfortunatelly, some parts of the water bodies are sometimes not included. However, you can manually edit the .xcm maps - .xcm file format is container - if you change it to .zip, you will see all included files of your map. You are looking for a water_area section. Map can be edited with OpenJUMP GIS program - you can fix whatever you need. Then change the .zip back to .xcm and have a fixed map. And please, if you fix it for yourself, share the improved map with community. Thank you

Go77, 2021-03-03:
Thank you MikeBravo15. I performed the actions you described and, yes, the area of concern is now colored blue. However, I suspect it's not just a map issue. In another issue I opened, I put forward the wish to be able to put waypoints over water area. What I see is that XCsoar still has waypoints over the water area that I coloured blue. So, it seems that there are other issues here than (just) the map issues.

MikeBravo15, 2021-03-05:
Sorry, I do not undrestand what the problem is - can you please upload a picture of the problem or the whole map? I believe, that XCSoar will put waypoints wherever you place them in waypoint file - there is no constraint. XCSoar is not a great tool for creating waypoints - you can create or edit them with SeeYou,  or some online editors (ie http://pottyplace.com/waypts/waypts.php or other), or simple text editor, if you are familiar with structure of .cup files - it is pretty simple. Waypoints and airspace areas are precisely defined via geographical coordinates and are acurate. Map export is somewhat less precise as there is some form of optimizing the shapes - XCSoar is gliding tool and map is for info only. However if you insist on precise representation of some map area, it can be achieved with OpenJump - I usually export small area of my interest from Openstreet maps (.osm file) as reference and open it together with XCSoar map - now you can modify XCSoar map elements pretty acurately according to the OpenstreetMap.

Go77, 2021-03-05:
Dear MikeBravo15. Yes, I know how to produce a .cup-file. I generated a square matrix with waypoints, each a few miles apart because I find the distribution of existing waypoints often way to coarse. And nowadays with GPS we no longer need bridges or highway-crossing as turn points. Here is a screenshot of my XCsoar:
[image] (Hope this works in this way in github, I'm not very experienced with github...). You can see the small circles representing my array of waypoints. They are absent in the water-areas although the waypoints are definitely there in the .cup-file! You can also see that waypoints are present in the water-area at the bottom, south of the Afsluitdijk. This is the part that I coloured blue, according to your suggestions. There are waypoints there, because XCsoar, I think, interprets this area as land.

MikeBravo15, 2021-03-06:
Interesting. If you disable map in setup, are all waypoints displayed correctly?

Go77, 2021-03-13:
No, very weird. Some are still displayed, others disappear. I propose to continue this discussion in my other item #530 'waypoints over water'.

NielsDav, 2021-05-04:
Hi @Go77 ,
I see you managed to succesfully edit the NL map. I face a similar problem in the Zeeland province in the Netherlands, were big parts are represented as land.
I am so far not succesful in editing the map in OpenJump GIS as suggested here earlier. Would you mind sharing your edited map for comparison?