XCSoar/XCSoar #536: Correct response to $POV,?,WL,BU

hpmax, 2021-03-04:
XCSoar versions having and not having the problem
-------------------------------------------------

XCSoar 7, older versions do not seem to support $POV,?,

Expected behavior
-----------------

XCSoar is sent: "$POV,?,WL,BU" it should respond with:

$POV,C,WL,xx
$POV,C,BU,xx

I'm not sure what xx should be if not yet set.

Actual behavior
---------------

$POV,C,NA,WL
$POV,C,NA,BU

I do not believe that "POV,C,NA..." is ever a valid response.

Steps to reproduce the behavior
-------------------------------

This was found during an enhancement request to Variod in OpenVario, it has not yet been published.  Sending the string mentioned in the title to XCSoar will re-create it however.

Do you have any idea what may have caused this?
-----------------------------------------------

I haven't looked through the code but this seems like a pretty simple straightforward fix where someone just coded the reply string incorrectly.

Do you have an idea how to solve the issue?
-------------------------------------------

See above.

MaxKellermann, 2021-03-04:
What is this and where is documentation on this protocol?

hpmax, 2021-03-04:
This allows applications like variod to request bugs and wing loading settings from XCSoar to ensure everything is synchronized.

The best answer I can give as to where it's documented is:
https://github.com/Openvario/meta-openvario/files/6014496/Extension.of.NMEA.Protocol.pdf

The $POV,?, is an important feature as without it, external varios which may be relying on XCSoar for polars or McCready info may get screwed up if rebooted.  However, the $POV,?,WL, and $POV,?,BU may be less important as they should be incorporated in the polars.  I do not know however if the format of the output is screwed up and it might issue an incorrect $POV,C,BU,xx if Bugs are manually entered into XCSoar.

lordfolken, 2021-03-13:
> ## XCSoar versions having and not having the problem
>
> XCSoar 7, older versions do not seem to support $POV,?,
> ## Expected behavior
>
> XCSoar is sent: "$POV,?,WL,BU" it should respond with: [...]
According to the linked document:
"If any of the requested values is unknown at the time of request, because the pilot hasn’t set the
value, then a Not-Available sentence will be sent instead."
$POV,C,NA,MC*27
So as long as you haven't setup wingloading, (configured a plane, set ballast) set an mc value or set bugs, this is the correct response.

hpmax, 2021-03-13:
Apparently I'm dyslexic.  I agree, XCSoar is complying with the spec.  It seems strange to me, but as long as it is compliant, it's fine.

closed 2021-03-13 by hpmax