XCSoar/XCSoar #537: Extra Maps option in Xcsoar

requa3r0, 2021-03-06:
Please consider to add another map option in XCSOAR, not based on elevations but actual maps for navigation, as an extra map option for visulatiation. 

The elevation map could still be l I added for elevation calculations in the background. 

As an example I would like to import the Danish Copenhagen Area ICAOas VFR map, to be used as a background in XCSOAR.

https://aim.naviair.dk/da/kort/

There are geo-referencerede geotiff or PDF 

It could be any other georeferenced map, as log as a topographical map can be included, and then the elevation map can be hidden. 

The elevation maps bringes almost no information, and the AGL can be calculated without it being displayed.

It would be great if you could make this possible.

lordfolken, 2021-03-25:
"It would be great if you could make this possible."
It is possible. Please code the feature and provide a pull request.

scls19fr, 2021-04-08:
About extra maps for XCSoar, you can also have a look at #521 about WMS and WMTS support.

scls19fr, 2021-10-07:
Some french maps can also be found as mbtiles files on https://site.glasow.fr/mbtiles-sia-france/

It could be nice if XCSoar could open that also