XCSoar/XCSoar #539: "Set as New Home" crashes XCSoar

Scumi, 2021-03-07:
XCSoar versions having and not having the problem
-------------------------------------------------
XCSoar @master branch

Expected behavior
-----------------

- Can "Set as New Home" on any airfield.

Actual behavior
---------------
Pressing "Set as New Home" instantly ends XCSoar. The following error message appears in the console:
`xcsoar: src/Widget/WindowWidget.cpp:81: virtual void WindowWidget::Hide(): Assertion 'window->IsVisible()' failed.`

Steps to reproduce the behavior
-------------------------------
- Compile XCSoar master
- Open XCSoar
- Select Waypoint on Map
- Click "Set as New Home"

Do you have any idea what may have caused this?
-----------------------------------------------
n/a

Do you have an idea how to solve the issue?
-------------------------------------------
n/a

bomilkar, 2021-03-08:
This might be related to issue #533 which was fixed a couple of days ago.

MaxKellermann, 2021-03-08:
True, it's the same bug, and it was fixed alreay.
And this bug report is bad because it does not name the exact commit id. Which is confusing, because this bug report was written *after* the bug was fixed.

closed 2021-03-08 by MaxKellermann

Scumi, 2021-03-08:
Hm, I re-compiled after pulling the fix yesterday, but still got the error. Will try again today evening.