XCSoar/XCSoar #549: download Site Files fails on Android

bomilkar, 2021-03-18:
XCSoar versions having and not having the problem
-------------------------------------------------
The issue appears in:
v7.0_preview17
ANDROID
git#e354c09bc
Dated March 16, 2021

I tried git#e354c09bc on UNIX and WIN64: both are OK.
Earlier versions of v7.0_preview ANDROID (end of Feb, 2021) used to work.

Expected behavior
-----------------

Config(1)->System->Site Files
gets you to the page with map file, airspace file, etc.
Click on one of these, press the "Download" button, select one of the files.
**The download will be done.**

Actual behavior
---------------

The file download start but never finishes.

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------
Side effect of updating some curl components??

Do you have an idea how to solve the issue?
-------------------------------------------

MaxKellermann, 2021-03-18:
I can't reproduce the problem.
CURL is not used for downloading files on Android.

bomilkar, 2021-03-18:
Hm...
I just updated my clone. Still the same issue. (The Info page now says version: 7.0_preview18-Android. But no git # anymore.)

I removed the "repository" file from the XCSoarData directory.
Even that file is not updated/downloaded on my Smartphone.

Config(1)->System->Site Files->More airspaces->Download->Download
A blank page is shown. Perhaps as a result of removing the "repository" file.
After a few seconds XCSoar crashes.

August2111, 2021-03-19:
I checked this issue with android too, XCSoar nightly build - preview 18! With download from a 'normal' http page I have no problems (f.e. waypoint and airspace files), but on https I got the error message 'Problem with the SSL CA cert (path? access rights?)' 
For example: Info->Weather->pc_met->RADAR->Deutschland
Obviously there is a certification error behind!

bomilkar, 2021-03-19:
It looks as if it depends on the smartphone I'm testing it on.
I have a tablet with Android version 8.0.0 which works and a smartphone with Android version 10 which doesn't.
Any idea where to dig deeper?

bomilkar, 2021-03-22:
In "adb shell" I did a screenrecord. Attached.
But all it does: demonstrates the issues exists. But I still don't see what happened inside. On Linux I'd use gdb to produce a backtrace. But here???

https://user-images.githubusercontent.com/20871255/111997360-443a9580-8b1b-11eb-915f-6971f4187190.mp4

MaxKellermann, 2021-03-25:
I tested with Android 9 (Cat S61) and Android 11 (Pixel 2), no problem. As long as I can't reproduce the problem, I can't help. Somebody else would need to debug this.

@August2111:
> Obviously there is a certification error behind!

This is not obvious to me. And I think you're wrong, because we're not using https here! Proof:
https://github.com/XCSoar/XCSoar/blob/d3ea258340d49b9cdf3be2bce34229321401cd61/src/Repository/Glue.cpp#L30

August2111, 2021-03-25:
Hi Max, you are right, this happens on the normal xcsoar download page without any SSL/TSL.
I'm working with high pressure on this topic - and I have the same effect, but with Android 6, 9 and 11 all things ok. Latest Monday I will give an update, I think this can be solved by me.

lordfolken, 2021-03-25:
presently stuff hosted on soaring.gahsys.com (airspace mainly) is unavailable as the host is down.

lordfolken, 2021-03-25:
i can confirm this on android 10. Maybe i can get a logcat.

bomilkar, 2021-03-26:
Using `logcat` is an excellent idea.
@lordfolken please share with us what it takes to enable `logcat` for XCSoar.
I'd like to help with debugging.

OBrown92, 2021-03-26:
I've tested this with Android 10 (Xiaomi Mi 9SE) and it's working fine.

bomilkar, 2021-03-26:
@OBrown92  Are you sure it's Android 10?
Looking at the spec on the web page it says Android 9.0

MaxKellermann, 2021-03-26:
@bomilkar enable the debug bridge on the phone, connect to computer, run "adb logcat" on the computer.

August2111, 2021-03-26:
My Run windows in the AVD emulator with Android 10 says:
[image]
This seem to be a permission problem coming with the change to platform-29 - only for Android-10. The previous platforms don't have this restriction and the android-11 is working well like expected. Every Google post says, this should be gone with Manifest setting 'android:requestLegacyExternalStorage="true"', but the issue is still there.

MaxKellermann, 2021-03-26:
@August2111 that exception dump is very helpful, because it means that hidden behind this bug, there's another bug in the native XCSoar code ignoring a Java exception from a JNI call.

MaxKellermann, 2021-03-26:
the SecurityException is now caught: 9d2c0a8e3c3a0573ce41635cd03c46267b5d1c1d
... but it's still there and causes download failures. At least no JNI error.

August2111, 2021-03-26:
I tried to move the data dir from ../XCSoarData to ../Documents/XCSoarData -- and this would solve this problem. From my point of view there is 'only' a 'small' bug in the development from android 9 to 11 ;-( - and it seems to be really a permission problem. Maybe there is another way to get the right permission to the data dir...

August2111, 2021-03-26:
My workaround is horrible for the system, because this needs a movement of all data (fortunately the data are readable) , but an other solution would be better :)
I tried a lot of things: different target platforms, different build tools - und now different NDK, but it helps nothing! 

The good thing: I learned a lot about android and the differences between the toolchains :)

lordfolken, 2021-03-26:
https://medium.com/microsoft-mobile-engineering/scoped-storage-in-android-10-android-11-28d58d989f3c

"4. Android 10 has lack of support for native libraries and file path. This is use-case for apps written in c, c++. This support has been added in Android 11, in-fact this is the only use case where developer is expected to write new code. Under the hood file path api is a proxy to media storage api. so if possible, call media store api directly."

August2111, 2021-03-26:
...this was my impression ;-(
The shortest solution for this seems to be my workaround: 
Moving the data directory from
- /storage/emulated/0/XCSoarData
to
- /storage/emulated/0/Documents/XCSoarData.
On this location I saw never this issue, all downloads are working fine

August2111, 2021-03-26:
From system side it is enabled the access to 'Documents' folder, so it is no need to set any additional permission

lordfolken, 2021-03-26:
So i guess the only fix is to adapt the path for android 10 only.

MaxKellermann, 2021-03-27:
We need to use the new storage APIs anyway, because Play will require targetSdkVersion=30 by August, and that version doesn't support the `requestLegacyExternalStorage` option.
This may be a good time to bring some order into XCSoarData (e..g subdirectories for different file types). I started some of that gradually, but never wanted to do completely due to our large legacy user base, and I wanted to avoid breaking anything.

bomilkar, 2021-03-27:
@August2111 I **cannot** confirm that this workaround fixes the issue:
> Moving the data directory from
> 
>     * /storage/emulated/0/XCSoarData
>       to
> 
>     * /storage/emulated/0/Documents/XCSoarData.
>       On this location I saw never this issue, all downloads are working fine

It creates an empty /storage/emulated/0/XCSoarData directory and starts from a fresh page.
Am I missing something??
I can also not confirm that there are any access rights preventing XCSoar from writing to /storage/emulated/0/XCSoarData. I see new xcsoar.log, Default.tsk, and *.prf files.

So it sounds as if @MaxKellermann suggestion is the way to go.
Would this be for 7.1 ? And leave 7.0 as is? That would be my recommendation.
Please let me know if I can be of any help applying the changes or testing (preferred) as it's implemented.

MaxKellermann, 2021-03-28:
New data point: a Nokia 3.1 with Android 10 doesn't reproduce the problem. Downloads work just fine.

OBrown92, 2021-03-28:
Sorry for late reply. @bomilkar, the mi9se got an update to Android 10. 
[image]
I've tested with latest Play Store testing version. Is this one effected or do I have to compile one?

MaxKellermann, 2021-03-28:
You don't need to compile. 7.0 is the latest.

OBrown92, 2021-03-28:
Okay just tested it again. Reinstalled everything and deleted the data folder. Everything works as aspected. If I can provide anything just let me know.

bomilkar, 2021-04-08:
I don't see it being fixed. At least not for Nokia's flavor of Android 10.
When I try to download for instance `data.fln` I get a message box with "Unsupported path".
I can start XCSoar w/o the XCSoarData directory. It then creates that directory and writes the usual files (xcsoar.log, Default.tsk, ..) to that directory. But when I try to download a Site File it complains "Unsupported path".
[image]

lordfolken, 2021-04-08:
Sorry closure was premature

benjarobbi, 2021-04-16:
Well I'm not sure if this is caused by this issue, however I can't download files on my new Redmi Note 10 Pro with Android 11. 

As soon I've downloaded a file using the file manager it immediately disappears after the download has finished.

lordfolken, 2021-04-16:
Does the app have storage permission?

bomilkar, 2021-04-18:
@benjarobbi  Which version are you running?
Info(3)->Credits
tell us: version, date, git.
Do you get any "Unsupported path" message?

schmidi, 2021-04-24:
Same problem here...

Android 10 on a Google Pixel (2016)
XCSoar from PlayStore
Version: 7.4-Android
Date: 8.4.2021

adb logcat shows 

`04-24 22:31:53.847  4015  6656 E DatabaseUtils: Writing exception to parcel
04-24 22:31:53.847  4015  6656 E DatabaseUtils: java.lang.SecurityException: Unsupported path /storage/emulated/0/XCSoarData/repository.tmp
04-24 22:31:53.847  4015  6656 E DatabaseUtils:         at com.android.providers.downloads.DownloadProvider.checkFileUriDestination(DownloadProvider.java:1085)
04-24 22:31:53.847  4015  6656 E DatabaseUtils:         at com.android.providers.downloads.DownloadProvider.insert(DownloadProvider.java:717)
04-24 22:31:53.847  4015  6656 E DatabaseUtils:         at android.content.ContentProvider$Transport.insert(ContentProvider.java:309)
04-24 22:31:53.847  4015  6656 E DatabaseUtils:         at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:154)
04-24 22:31:53.847  4015  6656 E DatabaseUtils:         at android.os.Binder.execTransactInternal(Binder.java:1021)
04-24 22:31:53.847  4015  6656 E DatabaseUtils:         at android.os.Binder.execTransact(Binder.java:994)`

I also tried to reinstall the app and removed the folder, did not help as mentioned in posts before...

henrik-the, 2021-04-30:
I have a similar problem:

Android 11 on Xiaomi Mi 11 Lite 5G
XCSoar from PlayStore
Version: 7.5
Date: 30.4.2021

I can select a File out of the Repository (e.g. germany-airspace)
The download progress goes from 0-100% but then I can't find the file in the XCSoarData Folder.

bomilkar, 2021-05-01:
@henrik-the , do you have a Documents folder? Maybe the file went in there?

henrik-the, 2021-05-01:
@bomilkar Yes I have the Documents Folder, but no file in there.

henrik-the, 2021-05-01:
Just out of curiosity i have compiled a version which uses the /emulated/0/Documents/ Path, but there i have the same behaviour.

XCSoar starts up normally and creates the folder and the standard files. With the Download Manager i can select a file, progress goes from 0-100% but no file gets saved...

I have also tested with a version, that includes some code to download files from WeGlide (which also uses the DownloadManager) and it also doesnt save a file.

lordfolken, 2021-05-13:
https://github.com/lordfolken/XCSoar/pull/new/downloadfix <- i've had some success by moving the xcsoardata to the application path.

bomilkar, 2021-05-14:
@lordfolken , what happens if XCSoar starts w/o any XCSoarData directory? Does it create one? Where?

MaxKellermann, 2021-05-14:
XCSoar itself does have access to XCSoarData. Just Android's DownloadManager refuses to put files in that directory. This isn't about XCSoar's own access.

lordfolken, 2021-05-15:
ok, so thx for @MaxKellermann's suggestion, i ripped out the android download manager and hence use curl like the other platforms.. This works like a charm, and to be honest its a lot simpler than the special case android download manager.

bomilkar, 2021-05-16:
This does the trick, at least on my phone running Nokia's flavor of Android 10. (I've tested it on Android 8.0: it still works.)
It may not be the fastest download on the planet, but that's not an issue.
Thanks, @MaxKellermann and @lordfolken ! I'd close this issue if you don't mind.

closed 2021-05-28 by MaxKellermann