XCSoar/XCSoar #550: Fast zoom-in on map crashes XCSoar

ubx, 2021-03-23:
XCSoar versions having and not having the problem
-------------------------------------------------
 The issue appears in:
XCSoar v7.0_preview18-Linux~git#40beee1184

The same problem is also in XCSoar v7.0_preview15-Linux

Expected behavior
---------------
Zoom into map.

Actual behavior
---------------
XCSoar crashed:
` xcsoar: src/ui/canvas/memory/Canvas.cpp:483: void Canvas::Stretch(PixelPoint, PixelSize, Canvas::ConstImageBuffer, PixelPoint, PixelSize): Assertion `dest_size.width < 0x4000' failed.`

Steps to reproduce the behavior
-------------------------------

1. Compile XCSoar:
make TARGET=UNIX OPENGL=n ENABLE_SDL=y USE_SDL2=y GEOTIFF=n

2. Start XCSoar: ./output/UNIX/bin/xcsoar -960x1280
3. Configuration > Site Files > Map database: ALPS_HighRes.xcm
4. Configuration > Look > Pages > Add: Main area:Map InfoBoxes:None, Bottom area:Nothing
5. Go to this Page
6. Zoom out to the max
7. Fast zoom in, i.e. keep the upper arrow key pressed
8. Repeat 6. and 7. until it crashes

Usually this may not be a problem, but if you have a rotary encoder which produces up/down arrow keys quite fast it is easy  to reproduce.

Do you have any idea what may have caused this?
-----------------------------------------------
Obviously it is the line `assert(dest_size.height < 0x4000)` in ui/canvas/memory/Canvas.cpp.

Do you have an idea how to solve the issue?
-------------------------------------------
I don't understand the mechanism behind this, but instead of crashing, why not just return, i.e. do nothing.

MaxKellermann, 2021-03-23:
Backtrace?

> Obviously it is the line assert(dest_size.height < 0x4000) in ui/canvas/memory/Canvas.cpp.

That line is correct. But the circumstances leading to this crash are not.

MaxKellermann, 2021-03-23:
Wait, I can reproduce it now.
Indeed 0x4000 as a limit was set more than 10 years ago when a high-res display was 480x640 and normal displays were 200x320. Pixel coordinates were 16 bit integers, and could easily overflow after a failed calculation, so all renderer calls must be very careful. That `assert()` should catch integer overflows.
These days, after abandoning WinCE, all pixel coordinates are 32 bit integers and will not overflow so easily. And displays are much larger now.
This thing was a reasonable calculation, but the pre-rendered map was zoomed so much that the destination area was larger than 0x4000 = 16k pixels (before clipping; after clipping, it was exactly the screen size).

closed 2021-03-23 by MaxKellermann

ubx, 2021-03-23:
Extremely fast fix, thanks a lot!