XCSoar/XCSoar #552: Infobox labels too small on XCSoar version 7

DanD222, 2021-03-25:
XCSoar version
--------------

v7.0_preview18

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

I find the infobox labels hard to read on XCSoar version 7 due to their small size. I feel that the infobox space could be better utilised to make the labels larger / taller in height.

I'm OK with the infobox label size as it is on v6.8 (even if it means some of the text is cropped - at least it doesn't require special effort to read it).

v7.0_preview18
[image]

v6.8.17
[image]

MaxKellermann, 2021-03-25:
XCSoar 7 auto-sizes those fonts so they fit a certain amount of text.
XCSoar 6.8 doesn't do that at all; it just uses some random scaling factor which is bad most of the time. Just look at how badly the text is truncated in most boxes and how bad the layout looks. Yes, the font is bigger, but that comes at a price.

Maybe the 7 algorithm needs some fine tuning.

What device are you using, what is the physical display size (in mm)? How large is the text (in mm)? How large was the text in 6.8? What size do you want it?

MaxKellermann, 2021-03-25:
I implemented a change which hopefully improves the problem for your specific case.

closed 2021-03-25 by MaxKellermann

DanD222, 2021-03-26:
Much better now.. thanks again!

[image]

MaxKellermann, 2021-03-26:
And much better than v6.8!