XCSoar/XCSoar #554: V7.0 display fails on Nexus 7 2nd Gen running r

mdw2021, 2021-03-26:
Which XCSoar version are you using when seeing the problem? V7.0
Which XCSoar version is the last one you know that did not show the problem? V 6.8.17
-->

Expected behavior
-----------------

Blank screen with only Quit box displayed.

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2021-03-26:
https://www.gsmarena.com/asus_google_nexus_7-4850.php

OS | Android 4.1.2 (Jelly Bean), upgradable to 5.1.1 (Lollipop)
-- | --
Chipset | Nvidia Tegra 3
CPU | Quad-core 1.2 GHz Cortex-A9
GPU | ULP GeForce

  | Announced | 2012, June. Released 2012, July
-- | -- | --
Status | Discontinued

Can you post the xcsoar.log please?
Also its a 9 year old device without any vendor support. please consider upgrading.

mdw2021, 2021-03-26:
Slightly different device running Android 6.0.1
https://www.gsmarena.com/asus_google_nexus_7_(2013)-5600.php
[2021-03-26T20:34:16Z] Starting XCSoar XCSoar v7.0-Android
Shader linker failed: --From Vertex Shader:
--From Fragment Shader:
Link was successful.

Shader linker failed: --From Vertex Shader:
--From Fragment Shader:
Link was successful.

[2021-03-26T20:34:16Z] Display dpi=320,322
[2021-03-26T20:34:16Z] OpenGL: mda=0 npot=1 stencil=0x88f0
[2021-03-26T20:34:16Z] Initialise fonts
[2021-03-26T20:34:28Z] Startup dialog
[2021-03-26T20:34:28Z] Loading profiles
[2021-03-26T20:34:28Z] Loaded profile from /storage/emulated/0/XCSoarData/default.prf
[2021-03-26T20:34:29Z] Display rotated
[2021-03-26T20:34:29Z] Initialise fonts
[2021-03-26T20:34:29Z] Initialise fonts
[2021-03-26T20:34:29Z] Loading language file
[2021-03-26T20:34:29Z] Loading input events file
[2021-03-26T20:34:29Z] OpenTerrain
[2021-03-26T20:34:29Z] Loading Topography File...
[2021-03-26T20:34:29Z] ReadWaypoints
[2021-03-26T20:34:29Z] Failed to read waypoint file: /storage/emulated/0/XCSoarData/user.cup
[2021-03-26T20:34:29Z] Failed to open 'airfields.txt' from ZIP file
[2021-03-26T20:34:29Z] Start at home waypoint
[2021-03-26T20:34:29Z] RASP load
[2021-03-26T20:34:29Z] No rasp data file: Failed to open ZIP archive /storage/emulated/0/XCSoarData/xcsoar-rasp.dat
[2021-03-26T20:34:29Z] ReadAirspace
[2021-03-26T20:34:29Z] Failed to parse airspace file: airspace.txt

[2021-03-26T20:34:29Z] Failed to open 'airspace.txt' from ZIP file
[2021-03-26T20:34:29Z] Register serial devices
[2021-03-26T20:34:29Z] ProgramStarted
[2021-03-26T20:34:29Z] Free storage 67108864 KB
[2021-03-26T20:34:29Z] cannot open /storage/emulated/0/XCSoarData/lua/init.lua: No such file or directory
[2021-03-26T20:34:50Z] Entering shutdown...
[2021-03-26T20:34:50Z] Free storage 67108864 KB
[2021-03-26T20:34:50Z] Saving profiles
[2021-03-26T20:34:50Z] Saving profile to /storage/emulated/0/XCSoarData/default.prf
[2021-03-26T20:34:50Z] Stop threads
[2021-03-26T20:34:50Z] Waiting for calculation thread
[2021-03-26T20:34:50Z] delete MapWindow
[2021-03-26T20:34:50Z] Save default task
[2021-03-26T20:34:51Z] Stop COM devices
[2021-03-26T20:34:51Z] Close Progress Dialog
[2021-03-26T20:34:51Z] Close Windows - main
[2021-03-26T20:34:51Z] Free storage 67108864 KB
[2021-03-26T20:34:51Z] Finished shutdown

groundhogxc, 2021-03-26:
This is a duplicate of #366, #386 and #466. The Nexus 7 (2013) has the same GPU as the Note 3, and the description (only Quit box visible) matches what is seen there.

We will see a few more of these I am afraid. Nexus 7 Gen 2 is a really nice device (good bright screen).

groundhogxc, 2021-03-26:
@lordfolken, you linked to the 2012 version of the Nexus 7 (which has a Nvidia Chipset). The Nexus 7 2013 has very different hardware,using the Adreno GPU. Not sure if you can edit your post to avoid confusion.

lordfolken, 2021-03-27:
@groundhogxc changed the entry to the correct device.

lordfolken, 2021-04-02:
@groundhogxc can you please test with 7.2?

mdw2021, 2021-04-02:
Issue resolved in Version 7.1

closed 2021-04-02 by mdw2021

groundhogxc, 2021-04-05:
@lordfolken sorry, did not see that immediately. I can also confirm this was fixed in commit [2451e25e55]. Thanks a lot!!!
(and, no, I do not have Paypal :-) )

lordfolken, 2021-04-05:
You should think @MaxKellermann for his time. :-)

groundhogxc, 2021-04-06:
That was indeed implied, it was his commit after all :-)
Anyway, strange that this bug only manifested itself in this exact GPU. Or potentially also in the one of issue #466 (which is some PowerVR variant).