XCSoar/XCSoar #555: Alternate list (simple mode): priority not respected

francescoRubini, 2021-03-29:
XCSoar versions having and not having the problem
-------------------------------------------------
last version (6.8.17-Android / Windows)

Expected behavior
-----------------
When i check alternate list (is set to "simple" mode) the first must be a reachable airport, then airfield

Actual behavior
---------------
in simple mode the prioriry is not respected. (the first is an airfield but there is an reachable airport)

Steps to reproduce the behavior
-------------------------------

For example i'm on this situation:
[image]

The first alternate is Brivio (an Airfield) not the airport (kong)
[image]

If I abort task then the alternates is correct:
[image]

lordfolken, 2021-04-05:
I think this isn't sufficiently explained in the docs or help. The task abort mode will always prioritize an airport over an out landing site if able. The alternates list when not it task abort, will show you the nearest landing possibility. The main idea is to show good airman-ship and implement a risk averse approach.

lordfolken, 2021-04-05:
Would you mind enhancing the help text and/or documentation in the manual?

francescoRubini, 2021-04-06:
ok.
I understand the mistake. In the italian version of application there is a error in help.

Now I check the english version.

[image]
 (sorted by reachable airport, then airfield and nearest)

francescoRubini, 2021-04-06:
I opened a pull request to correct the description.

I change english and Italian (also on weblate).

But in weblate there is the old key (https://hosted.weblate.org/translate/xcsoar/translations/en/?offset=1&q=The+alternates+will+only&sort_by=-priority%2Cposition&checksum=). I can't change this.

I hope my pull request is correct.