XCSoar/XCSoar #556: Status:Task Set MacCready value is not updated as soon as you change the value

MikeBravo15, 2021-03-30:
XCSoar versions having and not having the problem
-------------------------------------------------

XCSoar v7.0-Android , stable & develop

Expected behavior
-----------------

Calculation on the Status: task page are updated as soon as you change the value Set MacCready and close the selection dialog

Actual behavior
---------------

Calculated values remain the same, you have to change to other tab and then back to Task tab to update the values

lordfolken, 2021-04-02:
I see the issue, and it would be helpful to be just able to play with the mc value in order to see what is realistic. The values are only updated. 
Interestingly if you just change the value with keyboard navigation (left right with keyboard navigation on linux) it will update instantly.

MikeBravo15, 2021-08-14:
Recently I found, that several changes  of MCready values in a row triggers display of bogus values for estimated times and speed. Testing now with latest Android 7.12 version. I consider it a bug now

kobedegeest, 2026-01-14:
@coderabbitai that solution does not work please provide a plan for option one fixing the issue in TaskStatusPanel.cpp

kobedegeest, 2026-01-14:
@coderabbitai could you assist in writing the helper methode you mention in task 1.1

kobedegeest, 2026-01-14:
@coderabbitai I implemented your suggestion but issue still persist could you provide further hints/ insights

kobedegeest, 2026-01-15:
@lordfolken would the timer solution be an acceptable approach? It works i can clean up my commits and make a pr, the rabbit had some other suggestions forcing updates in task manager etc. but non of them worked.

With the 500ms delay you can visually see the values update so the timing seems fine, but it is not the most elegant solution.