XCSoar/XCSoar #560: Can't change AAT turn sector or start/finish line values.

RossDouglasMc, 2021-04-02:
XCSoar versions having and not having the problem
-------------------------------------------------

6.8

Expected behavior
-----------------

When changing type for turn points to Area Sector a values box should appear and I should be able to insert radius and radials. With Start and Finish lines I should be able to change the lenght, but the box with the values does not show.

Actual behavior
---------------

No value boxes come up when changing turn point types. I can't change the radius for a cylinder. I can't add radials or radius to an Area Sector.

Steps to reproduce the behavior
-------------------------------

Try to put in an AAT task with a different radius cylinder or Area Sector of 100km radius and 5 degrees either side of track. You can't do it. 

Do you have any idea what may have caused this?
-----------------------------------------------

Something is messed up in Version 7. It worked perfectly last week with version 6.8

Do you have an idea how to solve the issue?
-------------------------------------------

How do I revert to 6.8?

lordfolken, 2021-04-02:
@RossDouglasMc you can downloads old versions of xcsoar on http://download.xcsoar.org/releases/
In your description you are saying that 6.8 has the problem?

lordfolken, 2021-04-02:
@RossDouglasMc could you provide screenshots, and a better description of steps to reporduce?

RossDouglasMc, 2021-04-02:
Hi, No, 6.8 does not have the problem. In 6.8 when you select Change Type and then select an appropriate turn point, sector or start/finish line, underneath the Change Type box you then get dialogue boxes where you can change the  length of line, size of cylinder, or add radials to insert a wedge for AAT calculation. In V7 you don’t get these boxes appearing, so can’t adjust anything. Therefore the AAT functions can’t be used. It was all working for me perfectly on 6.8 two weeks ago. I came to the competition yesterday to find it had updated itself to the new 7.0 and I can’t use these functions. I am on an Android Samsung phone. I also have it on my laptop computer for use with Condor Soaring, but that has not updated and is still on 6.8 and working normally. RegardsRoss Sent from Mail for Windows 10 From: Philipp WollschlegelSent: Saturday, 3 April 2021 5:48 AMTo: XCSoar/XCSoarCc: RossDouglasMc; MentionSubject: Re: [XCSoar/XCSoar] Can't change AAT turn sector or start/finish line values. (#560) @RossDouglasMc you can downloads old versions of xcsoar on http://download.xcsoar.org/releases/In your description you are saying that 6.8 has the problem?

RossDouglasMc, 2021-04-02:
Hi, In case you didn’t get he last one,  XC SOAR V6.8 XC SOAR V7.0 Best regardsRoss Sent from Mail for Windows 10 From: Philipp WollschlegelSent: Saturday, 3 April 2021 5:49 AMTo: XCSoar/XCSoarCc: RossDouglasMc; MentionSubject: Re: [XCSoar/XCSoar] Can't change AAT turn sector or start/finish line values. (#560) @RossDouglasMc could you provide screenshots, and a better description of steps to reporduce?

lordfolken, 2021-04-03:
[image]

lordfolken, 2021-04-03:
Bisected to 89acd021dd827333ba41d19d75acec2505443cfe

lordfolken, 2021-04-03:
Actually its the settings per waypoint in any task type not just AAT.

closed 2021-04-04 by MaxKellermann

RossDouglasMc, 2021-04-05:
Thanks very much. Your service is outstanding. I note that we now have version 7.2 and it is all working perfectly.  Thanks again.Best regardsRoss McLernon Sent from Mail for Windows 10 From: Max KellermannSent: Sunday, 4 April 2021 9:33 PMTo: XCSoar/XCSoarCc: RossDouglasMc; MentionSubject: Re: [XCSoar/XCSoar] Can't change AAT turn sector or start/finish line values. (#560) Closed #560 via 7b57ce0.