XCSoar/XCSoar #563: Not seeing map/terrain files in picklist in v7.0

wsfreeman53, 2021-04-04:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2021-04-05:
Can you give us more info, please? 
- version of xcsoar (top of launch screen)
- version of android os

possibly related to #549 ?

lordfolken, 2021-04-06:
duplicate of #549

closed 2021-04-06 by lordfolken

wsfreeman53, 2021-04-07:
Initially did a download from xcsoar website of hi-res US_ILLINOIS_NO region. For some reason I ended up with a file that was called US_ILLINOIS_NO.xcm.zip. Since it looked like a zip file I unzipped it. That created a folder, not a single .xcm file, with the folder name US_ILLINOIS_NO.xcm,  with many files of different sorts within. I did not remember having to do that (i.e. unzipping) in the past. Later talked to another user and he confirmed it should be a single file with.xcm extension, not a folder. I guess that is why the xcsoar app did not appear in the pick list.
 I went back and did download of the standard res version for the same region, had no issues, and was able to assign that downloaded .xcm file to the xcsoar app with no issues. So my problem was solved, though I never resolved the exact source of it. Perhaps the hi-res version of the file is a zip file or got converted to a zip file somewhere along the way?

lordfolken, 2021-04-07:
So an xcm file is an archive (aka zip file) without compression. you can rename any .xcm to .zip file and unpack it. XCsoar will read everything even in unzipped format.
The double extension i guess is your OS that decided that that is a good idea. Possibly it recognized the mime type.