XCSoar/XCSoar #571: Kobo target build failing on v7.x

flightninja, 2021-04-07:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

MaxKellermann, 2021-04-07:
I don't see any compile error, because you did not cite any.

flightninja, 2021-04-07:
> I don't see any compile error, because you did not cite any.

Apologies, I have now updated the issue with error output.
@MaxKellermann if you are able to build successfully, it's possible it could be an issue only on my end...

closed 2021-04-07 by MaxKellermann