XCSoar/XCSoar #578: Fails to build on F-Droid

jspricke, 2021-04-10:
Hi,

your app started failing to build on F-Droid recently (since version 7.0):
```
configure: error: in `/home/vagrant/build/org.xcsoar/output/ANDROID7NEON/lib/arm-linux-androideabi/build/libsodium-1.0.18':
```
https://monitor.f-droid.org/builds/log/org.xcsoar/125

Would be great if you could help us fixing this.

lordfolken, 2021-04-10:
seems to be this:
https://gitlab.com/fdroid/fdroiddata/-/blob/master/metadata/org.xcsoar.yml

```

MaxKellermann, 2021-04-11:
> See `config.log' for more details

Where's the `config.log`?

@lordfolken is probably right, it's the wrong NDK version.

jspricke, 2021-04-11:
> * so this should use ./ide/provisioning/install-debian-packages.sh for setting up the environment
> * ./ide/provisioning/install-android-tools.sh would install the android environment
> * ndk version should be r22b and should be parsed from install-android-tools.sh

Thanks, I bumped the NDK version and got different errors. AFAIK it is currently not possible to automate them.
The next problem was that openssl-3.0.0-alpha13.tar.gz (references in v7.4) is not available anymore, so I replaced it by alpha14 but now I get:
```
crypto/ec/curve448/arch_64/f_impl.c:19:5: error: use of undeclared identifier 'uint128_t'
    uint128_t accum0 = 0, accum1 = 0, accum2;
    ^
crypto/ec/curve448/arch_64/f_impl.c:19:15: error: use of undeclared identifier 'accum0'
    uint128_t accum0 = 0, accum1 = 0, accum2;
              ^
crypto/ec/curve448/arch_64/f_impl.c:19:27: error: use of undeclared identifier 'accum1'
    uint128_t accum0 = 0, accum1 = 0, accum2;
                          ^
crypto/ec/curve448/arch_64/f_impl.c:19:39: error: use of undeclared identifier 'accum2'
    uint128_t accum0 = 0, accum1 = 0, accum2;
...
```

>   I might have to add that "we" never had f-droid as a release channel. Although i think we should, as its generally fitting with the philosophy of Free Software of the project.

F-Droid would be more then happy if you would use it as a release channel and help maintaining the build recipes.

lordfolken, 2021-04-11:
I'm also unable to build android at the moment due to the missing openssl library. Unrelated to f-droid. see:  https://github.com/lordfolken/XCSoar/runs/2316552632?check_suite_focus=true

linsui, 2021-05-18:
F-Droid failed to build 7.7 because the boost has switched away from bintray.

MaxKellermann, 2021-06-30:
The Boost URL was fixedb by be567726e13fb974f2c34fabec45b994012155d5

lordfolken, 2021-07-04:
Tried something: https://gitlab.com/lordfolken/fdroiddata/-/commit/242c569cd6d74c01a31cf341d9a8567e09f129e5 not sure why the linter is failing.

linsui, 2021-07-05:
https://gitlab.com/lordfolken/fdroiddata/-/jobs/1398385412#L252

lordfolken, 2021-08-10:
latest and coming builds should propagate now automatically.

closed 2021-08-10 by lordfolken