XCSoar/XCSoar #580: Broken link to actual version of openssl in build.py

fraktal-sb, 2021-04-11:
XCSoar versions having and not having the problem
-------------------------------------------------

OK  6.8.17
NOK  7.4

Expected behavior
-----------------

compile KOBO version should use the actual version of openssl

Actual behavior
---------------

compile error; missing link

[image]

Steps to reproduce the behavior
-------------------------------

make TARGET=KOBO TCSUFFIX=-8 DEBUG=n libs

Do you have any idea what may have caused this?
-----------------------------------------------

link in libs.py no mor available

[image]

[image]

Do you have an idea how to solve the issue?
-------------------------------------------

use this one

[image]

lordfolken, 2021-04-11:
sadly the openssl3.0.0alpha14 is broken for android.
https://github.com/openssl/openssl/issues/14804
potential fix:
https://github.com/openssl/openssl/pull/14812

lordfolken, 2021-04-19:
https://github.com/XCSoar/XCSoar/commit/58bd0576101d9008925a89fcbb70839e30f6ee3b Updates the ssl lib to current 3.0.0alpha14. Still broken though.

MaxKellermann, 2021-04-19:
Works here.

CazYokoyama, 2021-04-26:
https://www.openssl.org/source/ has alpha15 and no longer has 14.

lordfolken, 2021-05-10:
fixed in 5743209f4db78eacfec0cfe51b2231cfae680387

closed 2021-05-10 by lordfolken