XCSoar/XCSoar #581: Kobo Version compile error: unknown type name uint128_t

fraktal-sb, 2021-04-11:
XCSoar versions having and not having the problem
-------------------------------------------------

OK      v7.0_preview_15
NOK   v7.4

Expected behavior
-----------------

make TARGET=KOBO TCSUFFIX=-8 DEBUG=n libs

should complete without errors

Actual behavior
---------------

compile error

crypto/ec/curve448/arch_64/f_impl.c:19:5: error: unknown type name 'uint128_t'; did you mean 'uint32_t'?
     uint128_t accum0 = 0, accum1 = 0, accum2;
     ^~~~~~~~~

Steps to reproduce the behavior
-------------------------------

make clean
make TARGET=KOBO TCSUFFIX=-8 DEBUG=n libs

operating environment (uname -r) Debian 4.19.0-13-amd64
[image]

Do you have any idea what may have caused this?
-----------------------------------------------

No

Do you have an idea how to solve the issue?
-------------------------------------------

No

fraktal-sb, 2021-04-15:
I solved this by switching from gcc-8/g++8 to gcc-10/g++10

Is this the solution?

MaxKellermann, 2021-04-15:
No.
Are you sure the XCSoar master branch ever had this problem? I assume you manually changed the OpenSSL version to 3.0.0-alpha14, but didn't adjust the Configure flags!
See https://github.com/XCSoar/XCSoar/commit/58bd0576101d9008925a89fcbb70839e30f6ee3b

fraktal-sb, 2021-04-15:
I also tried with

https://ftp.wrz.de/pub/openssl/source/openssl-3.0.0-alpha13.tar.gz

same problem. always used a fresh buster.

with alpha14 it was a bit better (libs compiled with gcc/g++-8) but then in the next step other problems. 

does gcc-8 have problems with 128 bit data types (see error message)?

the version generated with gcc/g++-10 I also tested on my GLO and it seems to work

fraktal-sb, 2021-04-16:
error messages with latest master
[image]

MaxKellermann, 2021-04-16:
XCSoar downloads the libstdc++10 headers, but you're using an older GCC version.
This is a mess currently, because XCSoar uses this workaround because it wants to use Musl, but the cleanest solution would be to build a full compiler toolchain during the XCSoar build.

fraktal-sb, 2021-04-16:
no problem, the "full solution" can wait. We have the 7.5 version people can work/test with and maybe things ressolve when we have the new compiler in the next bullseye version.

MaxKellermann, 2021-05-04:
> the cleanest solution would be to build a full compiler toolchain during the XCSoar build.

Voila: https://github.com/XCSoar/XCSoar/commit/9582c34e01291cca222b6b30713fcc2fa17ae4cd

fraktal-sb, 2021-05-04:
tested with:

make TARGET=KOBO DEBUG=n libs

got message:

make: /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-g++: Kommando nicht gefunden

something more necessary?

MaxKellermann, 2021-05-04:
Try a clean Kobo build first (`make TARGET=KOBO clean` and then try again).

fraktal-sb, 2021-05-04:
did not help;
tried on 2 different machines (32/64 bit) with fresh Debian 10 and the following base setup:

sudo apt-get install git
sudo apt-get install build-essential
git clone git://github.com/XCSoar/XCSoar
cd XCSoar

git submodule init
git submodule update
dpkg-checkbuilddeps
sudo apt-get install libegl1-mesa-dev libsdl2-dev libfreetype6-dev libgl-dev libgl1-mesa-dev libpng-dev libjpeg-dev libsodium-dev libtiff5-dev libtiff-dev libgeotiff-dev libasound2-dev libc-ares-dev liblua5.2-dev librsvg2-bin imagemagick xsltproc ffmpeg

make -j5 TARGET=KOBO DEBUG=n libs

lordfolken, 2021-07-04:
Please use https://github.com/XCSoar/XCSoar/blob/master/ide/provisioning/install-debian-packages.sh to setup your environment. Another option would be to pull the docker container and compile the KOBO version with xcsoar-compile KOBO.

closed 2021-07-04 by lordfolken

fraktal-sb, 2021-07-05:
sorry about asking again:

After using install-debian-packages.sh I still have the the following messages (on a new clean debian 10 system)

make -j5 TARGET=KOBO TCSUFFIX=-8 DEBUG=n libs

make: /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-g++-8: Kommando nicht gefunden
./build/thirdparty.py ./output/KOBO UNIX arm-linux-gnueabihf armv7a-a8neon-linux-musleabihf " -march=armv7-a -mfpu=neon -mfloat-abi=hard -mthumb -fno-PIE" "-I./output/KOBO/include -DHAVE_POSIX -DHAVE_STDINT_H -DHAVE_UNISTD_H -DHAVE_VASPRINTF -DKOBO -DDITHER -DGREYSCALE " "--static -Wl,-u,pthread_cond_signal -Wl,-u,pthread_cond_broadcast -Wl,-u,pthread_cond_wait" /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-gcc-8 /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-g++-8 /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-ar "rcs" /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-ranlib /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-strip "/home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-windres"
download https://ftp.gnu.org/gnu/binutils/binutils-2.36.1.tar.xz
checking build system type... i686-pc-linux-gnu
checking host system type... i686-pc-linux-gnu
checking target system type... armv7a-a8neon-linux-musleabihf
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... yes
checking for a sed that does not truncate output... /usr/bin/sed
checking for gawk... no
checking for mawk... mawk
checking for gcc... ccache gcc
checking whether the C compiler works... no
configure: error: in `/home/josef/XCSoar/output/KOBO/lib/build/binutils-2.36.1':
configure: error: C compiler cannot create executables
See `config.log' for more details
Traceback (most recent call last):
  File "./build/thirdparty.py", line 203, in <module>
    x.build(toolchain)
  File "/home/josef/XCSoar/build/python/build/gcc.py", line 8, in build
    AutotoolsProject.build(self, toolchain.native, target_toolchain=toolchain)
  File "/home/josef/XCSoar/build/python/build/autotools.py", line 111, in build
    build = self.configure(toolchain, target_toolchain=target_toolchain)
  File "/home/josef/XCSoar/build/python/build/autotools.py", line 98, in configure
    subprocess.check_call(configure, cwd=build, env=toolchain.env)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/josef/XCSoar/output/src/binutils-2.36.1/configure', 'CC=ccache gcc', 'CXX=ccache g++', 'CFLAGS=-Os -ffunction-sections -fdata-sections -fvisibility=hidden', 'CXXFLAGS=-Os -ffunction-sections -fdata-sections -fvisibility=hidden', 'CPPFLAGS=-DNDEBUG ', 'LDFLAGS= ', 'LIBS= ', 'AR=ar', 'ARFLAGS=', 'RANLIB=ranlib', 'STRIP=strip', '--prefix=/home/josef/XCSoar/output/KOBO/lib', '--enable-silent-rules', '--target=armv7a-a8neon-linux-musleabihf', '--with-system-zlib', '--enable-gold', '--disable-ld', '--disable-libquadmath', '--disable-lto']' returned non-zero exit status 77.
make: *** [build/thirdparty.mk:45: output/KOBO/lib/stamp] Fehler 1

MaxKellermann, 2021-07-05:
Don't use TCSUFFIX. Our build scripts build a full ARM cross toolchain during the XCSoar build, and specifying TCSUFFIX overrides that and chooses your broken toolchain.

fraktal-sb, 2021-07-05:
still no change on fresh/clean Debian with fresh install-debian-packages.sh (i386).

Is x64 required?

make TARGET=KOBO DEBUG=n

make: /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-g++: Kommando nicht gefunden
./build/thirdparty.py ./output/KOBO UNIX arm-linux-gnueabihf armv7a-a8neon-linux-musleabihf " -march=armv7-a -mfpu=neon -mfloat-abi=hard -mthumb -fno-PIE" "-I./output/KOBO/include -DHAVE_POSIX -DHAVE_STDINT_H -DHAVE_UNISTD_H -DHAVE_VASPRINTF -DKOBO -DDITHER -DGREYSCALE " "--static -Wl,-u,pthread_cond_signal -Wl,-u,pthread_cond_broadcast -Wl,-u,pthread_cond_wait" /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-gcc /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-g++ /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-ar "rcs" /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-ranlib /home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-strip "/home/josef/XCSoar/output/KOBO/lib/bin/armv7a-a8neon-linux-musleabihf-windres"
download https://ftp.gnu.org/gnu/binutils/binutils-2.36.1.tar.xz
checking build system type... i686-pc-linux-gnu
checking host system type... i686-pc-linux-gnu
checking target system type... armv7a-a8neon-linux-musleabihf
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... yes
checking for a sed that does not truncate output... /usr/bin/sed
checking for gawk... no
checking for mawk... mawk
checking for gcc... ccache gcc
checking whether the C compiler works... no
configure: error: in `/home/josef/XCSoar/output/KOBO/lib/build/binutils-2.36.1':
configure: error: C compiler cannot create executables
See `config.log' for more details
Traceback (most recent call last):
  File "./build/thirdparty.py", line 203, in <module>
    x.build(toolchain)
  File "/home/josef/XCSoar/build/python/build/gcc.py", line 8, in build
    AutotoolsProject.build(self, toolchain.native, target_toolchain=toolchain)
  File "/home/josef/XCSoar/build/python/build/autotools.py", line 111, in build
    build = self.configure(toolchain, target_toolchain=target_toolchain)
  File "/home/josef/XCSoar/build/python/build/autotools.py", line 98, in configure
    subprocess.check_call(configure, cwd=build, env=toolchain.env)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/josef/XCSoar/output/src/binutils-2.36.1/configure', 'CC=ccache gcc', 'CXX=ccache g++', 'CFLAGS=-Os -ffunction-sections -fdata-sections -fvisibility=hidden', 'CXXFLAGS=-Os -ffunction-sections -fdata-sections -fvisibility=hidden', 'CPPFLAGS=-DNDEBUG ', 'LDFLAGS= ', 'LIBS= ', 'AR=ar', 'ARFLAGS=', 'RANLIB=ranlib', 'STRIP=strip', '--prefix=/home/josef/XCSoar/output/KOBO/lib', '--enable-silent-rules', '--target=armv7a-a8neon-linux-musleabihf', '--with-system-zlib', '--enable-gold', '--disable-ld', '--disable-libquadmath', '--disable-lto']' returned non-zero exit status 77.
make: *** [build/thirdparty.mk:45: output/KOBO/lib/stamp] Fehler 1

fraktal-sb, 2021-07-16:
problem solved: module ccache (which is not part of standard debian) installed; this module should be integrated in https://github.com/XCSoar/XCSoar/blob/master/ide/provisioning/install-debian-packages.sh

lordfolken, 2021-07-18:
Thanks for your contribution. You do not need ccache to build anything. Its a caching mechanism for repetitive builds that is entirely optional. Does your home directory and shell perhaps specify CC=ccache g++ or similar in your .bashrc or .profile?

Might i suggest you create a new user and try compiling there?

lordfolken, 2021-07-18:
Going through the build targets ... this might be the issue:  

 https://github.com/XCSoar/XCSoar/blob/9be4ec5738529814aa6a5c7a2dfff44b8cbfd760/build/thirdparty.py#L46