XCSoar/XCSoar #588: Unreadable german labels in main menus

fraktal-sb, 2021-04-16:
XCSoar versions having and not having the problem
-------------------------------------------------

all

Expected behavior
-----------------

2 german labels are unreadable and should be corrected (how to see screenshot from Kobo Glo-HD). Correct labels do not irritate and are better for overall acceptance of the software

Actual behavior
---------------

Labels are unreadable (see screenshot)

Steps to reproduce the behavior
-------------------------------

go into menus pan and config (2/3)

Do you have any idea what may have caused this?
-----------------------------------------------

not enough space for non separated labels

Do you have an idea how to solve the issue?
-------------------------------------------

the patch below should correct this in directly in file de.po

[image]

diff --git a/po/de.po b/po/de.po
index ad00fadfa3..9047e8a071 100644
--- a/po/de.po
+++ b/po/de.po
@@ -6074,7 +6074,9 @@ msgstr "Wegpunkt löschen?"
 
 #: src/Dialogs/Waypoint/Manager.cpp:295 Data/Input/default.xci:597
 msgid "Waypoints Editor"
-msgstr "Wegpunktbearbeitung"
+msgstr ""
+"Wegpunkt-\n"
+"bearbeitung"
 
 #: src/Dialogs/Waypoint/Manager.cpp:304
 msgid "Save changes to waypoint file?"
@@ -10358,7 +10360,8 @@ msgid ""
 "Pan\n"
 "Off"
 msgstr ""
-"Verschieben\n"
+"Ver-\n"
+"schieben\n"
 "Aus"
 
 #: Data/Input/default.xci:48

MikeBravo15, 2021-04-17:
I guess, that .po edits are better solved via Weblate https://hosted.weblate.org/projects/xcsoar/translations/

fraktal-sb, 2021-04-17:
translation entered in weblate, but automatic quality control has problems with. let's see what will be happen

[image]

MikeBravo15, 2021-04-17:
Just confirm the translation by saving it again. You can test the results imediately, if you export the translation to .mo file, put it into XCSoar main directory and the select it in Config; Look; Language, Input

fraktal-sb, 2021-04-18:
done, but translation export was only possible in .po format. The .mo file I generated the normal way by compiling the .po file
[image]

MikeBravo15, 2021-04-18:
In my case I use menu Dateien, there you can select the format of the output file