XCSoar/XCSoar #590: Request confirmation on pressing "Revert task"

VladimirVLF, 2021-04-19:
After defining a task in XCSoar on my smartphone I accidentally touched the "Revert task" button and did no notice it.
Then I sent task declaration to my logger.
Luckily I looked at the declaration in my logger (which I usually don't do) and found that my task is reverted.
After going back and forth with XCSoar and the logger and, I realized that my task in XCSoar is reverted, which is not noticeable at a glance, since task legs look the same.

It would be a good idea to ask for a confirmation when pressing "Revert task" when defining the task.

MaxKellermann, 2021-04-20:
Is this about Revert Task or Reverse Task?

lordfolken, 2021-05-01:
I think he meant reverse. @VladimirVLF please provide a pr for inclusion.

MaxKellermann, 2021-06-30:
No reply.

closed 2021-06-30 by MaxKellermann