XCSoar/XCSoar #595: CreateContext error

doramail, 2021-04-29:
XCSoar versions having and not having the problem
-------------------------------------------------
All versions post 6.8.0.17 are having the problem on my TECLAST P80X Phablet.
It is an ANDROID version 9 Model: P80X_EEA (G5K1) with 2.0 GB of RAM
CPU: SC9863A
FLASH:32 GB.

Expected behavior
-----------------
Program execute correctly.

Actual behavior
---------------
Installing any version beginning with 7 produce the same behavior after launch of the application.
When launching CXSOAR I have this error message displayed :

"org.xcsoar.EGLException: eglCreateContext() failed: 12293"

I have to kill the application to exit.

Steps to reproduce the behavior
-------------------------------
The update process of version 6.8.17 is OK the problem arise only when trying to run the newly installed versions. "7."

Do you have any idea what may have caused this?
-----------------------------------------------
Absolutely not.

Do you have an idea how to solve the issue?
-------------------------------------------
No

MaxKellermann, 2021-04-29:
This is a graphics driver problem and can only be analysed and fixed if a developer has such a device.

doramail, 2021-04-29:
Thank you very much for your very fast answer.
Your answer supposes that you have made some modifications to the way XCSoar uses the graphics driver ?
I can still use the old version 6.8.17 for the moment, which has not this graphics driver issue.
Best regards,

MaxKellermann, 2021-04-29:
XCSoar up to 6.x used OpenGL/ES 1.0 for graphics output, and XCSoar 7.x has updated to OpenGL/ES 2.0. Those OpenGL/ES versions are very different, and they employ very different parts of the graphics driver. That change may have caused the problem, and to fix it, a deeper analysis is needed.

doramail, 2021-04-29:
Thanks a lot for these informations.
Do you think an upgrade from Android 9 Version could solve this anoying issue ?

MaxKellermann, 2021-04-29:
Maybe. If this is a graphics driver bug, it may be fixed in the new version. If this is a XCSoar bug, it may not be fixed. Right now, I can't tell more, I can only guess randomly.

doramail, 2021-04-29:
OK, I will try to upgrade, I will let you know if it has solved the problem.
Thank you, hope the issue is not XCSoar.

DigiOrti, 2021-05-04:
Same problem here. Teclast T10 shows the same exception. Doesn't work anymore.

doramail, 2021-05-25:
Hi, I finaly installed Android version 11 on my Teclast P80X (G5K6) and the problem is still present. It means, if I am not wrong, that the createcontext error is due to hardware issue and has nothing to do with Android version. That is sad since I cannot use my tablet anymore with the new XCSoar versions.
Best regard.

MaxKellermann, 2021-05-26:
"device needed".
What if those interested in getting this fixed collect donations and use it to buy the hardware for XCSoar developers?
Maybe we need a central way to donate for test hardware?

doramail, 2021-05-27:
I totally agree, we just have to find a sufficient number of interested persons, 10 maybe 20.