XCSoar/XCSoar #599: XCSoar 7.6

weemacgregor, 2021-05-01:
Greetings Gents:

Thanks for all the work in producing this outstanding app.

I just downloaded 7.6 to my Samsung Galaxy Tab running Android 5.1.1.

Now when it is time to load airspace files from within XCSoar with the 'Config' tab using the File Manager, no files appear, and I cannot add any;  it appears that XCSoar is no longer connecting to the internet.  When I uninstalled and re-Installed XCSoar, I noticed that the initial dialog window (which opens up when I install XCSoar from the Google Play app) does not ask permission to use the Wifi connection..... it only asks permission to use Location, Photos/Media/Files, and Bluetooth connection information.....  that's all.  I did not have this issue running 7.4.

lordfolken, 2021-05-02:
Can you post the xcsoar log from the directory? 
It asks for this permission:
  <uses-permission android:name="android.permission.INTERNET"/>

Does 6.8.X work? https://download.xcsoar.org/releases/ for old releases.
https://github.com/XCSoar/XCSoar/blob/master/android/AndroidManifest.xml

NielsDav, 2021-05-04:
Hi guys,

I have the same issue as @weemacgregor on version 7.6, also when trying XCSoar 7.4 and 7.5.
Not sure if I already had this issue when running 7.4 and 7.5 before though.
Running Android 10; see screenshot and log.

[xcsoar.log](https://github.com/XCSoar/XCSoar/files/6419517/xcsoar.log)

[image]

When running XCsoar 6.8.17, it works fine again.

lordfolken, 2021-05-04:
This bug report is for android 5. The issue you are refeering to is #549. @NielsDav

lordfolken, 2021-05-04:
And this seems to be a regression as it worked on 7.4. @MaxKellermann

MaxKellermann, 2021-05-04:
Or maybe it never worked?
7.4 brought the error message. Previously, errors were invisible/ignored.

weemacgregor, 2021-05-05:
I have a correctly working copy of XC Soar 6.8.15 working on a Dell Streak (Android 2.3.7).  I copied the repository executable file from the XCSoarData folder on this Dell, and copied it to the XCSoarData folder on the Samsung Galaxy Tab running Android 5.1.1 with XCSoar 7.6.  Now, XCSoar will find and download airspace files from the internet, and puts the airspace file in the correct folder, but it appends a 'tmp' file extension to the xcm file - this tmp extension seems to prevent me from accessing the terrain file from within XCSoar.  As soon as I use a file manager to remove the 'tmp' extension, XCSoar recognizes the file and loads the terrain.

So for now, my 'workaround' is to:
1. Transfer a working copy of the repository Unix executable file to my troublesome version of XCSoar7.6 on my Samsung Galaxy Tab running Android 5.1.1
2. Download the required airspace and terrain files to the troublesome version of XCSoar on Android 5.1.1.
3. Use Android's file manager to remove the 'tmp' file extension.

Things now work as expected.
[image]
[image]

lordfolken, 2021-06-09:
this was fixed with https://github.com/XCSoar/XCSoar/commit/0fc220cd7b57ad42273a130b0813ea0f7581a2c9

MaxKellermann, 2021-06-09:
@lordfolken this restriction doesn't exist on Android 5, are you sure this is fixed?

lordfolken, 2021-06-09:
@MaxKellermann the tmp extension was the issue when the file was being moved post download between the two folders. One was a fuse mount the other sdcardfs. Samsung is the original implementer of sdcardfs. 

@weemacgregor could you test again with 7.8?

weemacgregor, 2021-06-10:
Greetings;  it will be my pleasure to check my set-up with 7.8.  Please give me a few days, as I am away from home for the next 3 days.  Will report back.
Thanks.

weemacgregor, 2021-06-16:
Greetings Gentlemen:

I updated my Samsung Galaxy Tab running Android 5.1.1 with XCSoar 7.8 from the Google Play store.
Good news and bad news (at least for me):

Did a fresh install of XCSoar 7.8

1.	XCSoar 7.8 installed correctly, including asking for access to SD card (see figure 1).  Access is of course given, and is further confirmed by the Android OS (see figure 2).

2.	When using the built-in XCSoar File Manager to download Map Database, Waypoint, and Airspace files, the list of available downloads appears correctly.  When I select one to download, it queues correctly and then downloads itself.

3.	When the downloaded file is then loaded into XCSoar, it displays correctly.

So far so good.

However:

There are problems with the location in which XCSoar stores the dowloaded data (Map Databases, Waypoint files, and Airspace files):

1.	XCSoar seems to have put all downloaded files into a XCSoarData folder which it creates  on the device itself, instead of the SD card (see figures 3 and 4). 

2.	This folder, residing at :  /storage/emulated/0XCSoarData, was not created during the initial installation of XCSoar - I looked - but was created after I started up XCSoar and used the built in File Manager to download the usual Map Database, Airspace and Waypoint files.

So….  for myself, one problem was solved:  the built-in File Manager is now working to retrieve terrain, airspace and waypoint files.

But….  these data files now reside on the device itself, and any  Map Database, Waypoint or Airspace file, previously loaded onto the SD card, is completely ignored.

Incidentally, to check this last point, I removed XCSoar 7.8,  and then used my computer to write Map Database, Waypoint  and Airspace files into an XCSoarData folder which I had written to my SD card.  When I re-installed XCSoar 7.8 and despite permission given for XCSoar to access and modify the SD card, it completely ignored any data written to the XCSoar folder residing on the SD card.

[image]
[image]
[image]
[image]

lordfolken, 2021-06-16:
Great to hear that downloads work again. 
This issue is probably a regression due to:
XCSoar/XCSoar/commit/be2ba8c06f60e55911c7ae32c280ee5382fc98cd

We had some hacks for Samsung devices that used other locations than the default location /storage/emulated/0 of the sdcard. 

@MaxKellermann what do you think?

MaxKellermann, 2021-06-16:
XCSoar should enumerate all storages and look for XCSoarData in all of them; just like XCSoar searched all drive letters on WinCE.
When those Samsung specific kludges were written, I think those enumeration methods did not exist yet in Android. I don't want to bring those kludges back.

weemacgregor, 2021-06-16:
Greetings again gents:
I do not wish to make more work for you guys;  as I said initially, you have a great product, and I am certainly not complaining:  but please note that with my same Samsung hardware/software combination, XCSoar 7.6 was able to read data on the SD card - XCSoar 7.8 does not seem to.....

lordfolken, 2021-06-16:
As far as i gather from some googleing: 

"Environment.getExternalStorageDirectory() refers to whatever the device manufacturer considered to be "external storage". On some devices, this is removable media, like an SD card. On some devices, this is a portion of on-device flash. Here, "external storage" means "the stuff accessible via USB Mass Storage mode when mounted on a host machine" 

So the function does not garantee that the return will be a removable sdcard.

I found this code snippet to check whether its really a removable sdcard:
https://gist.github.com/PauloLuan/4bcecc086095bce28e22#gistcomment-2591001

I also foundout that there are some hidden functions int he sdk... 
from:
https://github.com/liuxu0703/MyAndroidStaleCode/blob/master/app/src/main/java/com/lx/mystalecode/utils/StorageManagerHack.java

What we could do is enumerate the storage paths, use the above function to see whether its an sdcard, and then use it. 

However i read somewhere that using the StorageManager functions is breaking change and deprecated from api level 28 on.

lordfolken, 2021-06-16:
> Greetings again gents:
> I do not wish to make more work for you guys; as I said initially, you have a great product, and I am certainly not complaining: but please note that with my same Samsung [...]

7.7 and 7.8 was released due to downloads no longer functioning on android 10. #549 in the process we removed those special cases for samsung devices. Which is why XCSoarData is now found on your "external" storage instead of your sdcard.

weemacgregor, 2021-06-16:
7.7 and 7.8 was released due to downloads no longer functioning on android 10. #549 in the process we removed those special cases for samsung devices. Which is why XCSoarData is now found on your "external" storage instead of your sdcard.

No worries:  for the time being, I think (I have not tried yet) I can use the Android file manager to move files from my SD card to the XCSoarData folder that XCSoar recognizes and uses - it's just an extra manual step on my part.

MaxKellermann, 2021-08-23:
So looks like the download problems are "solved" (as in: this was never a problem). The rest of this bug report confuses me, I don't understand what it means, but since this isn't about a problem with the download manager, it doesn't matter here. If another problem still exists, write a new bug report.

closed 2021-08-23 by MaxKellermann