XCSoar/XCSoar #600: Broken links for XCSoar 7.0 and the users forum

Flying-Roger, 2021-05-02:
Hello, I do not know how I should report this kind of issue.
Most of the links on the https://www.xcsoar.org/ page are broken:
- For example, the "Manual for XCSoar 7.6": https://download.xcsoar.org/releases/7.6/XCSoar-manual.pdf  
   returns "The requested URL was not found on this server."

The same problem occurs for all the other links on page https://www.xcsoar.org/discover/manual.html

When trying to access the forum, I get another error
=> Forbidden
      You don't have permission to access this resource.
      Apache/2.4.38 (Debian) Server at forum.xcsoar.org Port 443

lordfolken, 2021-05-02:
We are aware, the docs are currently missing.

Flying-Roger, 2021-05-02:
OK to rename this post for the documents link.
However the forum link issue is not really a "missing document"

Flying-Roger, 2021-05-02:
OK.
No problem for me as I found the old docs on the repository, but visitors might be surprised to find most of the links broken.

What about the access denial to the xcsoar forum?

 

Roger

lordfolken, 2021-05-02:
fixed in https://github.com/XCSoar/XCSoar/commit/a1cd6a284cf4340e0e91b855c7eb0829f7cc7255

The forum access issue is another problem, that is mainly due to spam filtering, its not really down. Please come to irc://irc.freenode.org#xcsoar and talk to apf

closed 2021-05-02 by lordfolken