XCSoar/XCSoar #601: Cross Section label (height dimension) unreadable

fraktal-sb, 2021-05-02:
XCSoar versions having and not having the problem
-------------------------------------------------

7.5

Expected behavior
-----------------

Label should not overwrite axis values

Actual behavior
---------------

the height label overwrites the highest axix value (see screenshot)

[image]

Steps to reproduce the behavior
-------------------------------

Go to Cross Section

Do you have any idea what may have caused this?
-----------------------------------------------

label text does not clip, in the distance axis (see screenshot) the problem seems been solved

Do you have an idea how to solve the issue?
-------------------------------------------

label text should clip text below

MaxKellermann, 2021-05-03:
> XCSoar versions having and not having the problem
> 7.5

This is confusing.

fraktal-sb, 2021-05-03:
versions having and not having the problem:

6.8.17 was a different layout
bug is in the newest master

behavior in ./Renderer/ChartRenderer.cpp ChartRenderer::Finish
is strange: when changing there from SetBackgorundTransparent to SetBackgroundOpaque

it seems to work correctly (see screenshot).  x- and y- labels are correct. When using "Transparent" only x- label is correct

actualy did not find the reason

[image]

fraktal-sb, 2021-05-04:
the following solution is proposed (UNIX tested); see patch and screenshots

[image]
[image]

[patch.txt](https://github.com/XCSoar/XCSoar/files/6419871/patch.txt)

MaxKellermann, 2021-05-04:
@fraktal-sb you should submit patches as PR, not by attaching a text file inside an issue comment.
However the change isn't good; if texts overlaps, you still draw both, but you also draw a white background rectangle, which "undraws" the lower text. That adds more overhead.
Better would be to clip texts properly; if texts overlap, don't draw them. That not only fixes this problem, but also reduces overhead.