XCSoar/XCSoar #602: resize window in WIN64 exits with runtime assertion

bomilkar, 2021-05-02:
XCSoar versions having and not having the problem
-------------------------------------------------
Information from the Credits page:
version: 7.6-PC
date: May 2 2021
git: ba1a6db0f7

Expected behavior
-----------------
Changing the size of the Window within XCSoar is running should not crash.

Actual behavior
---------------
Making the window wider/narrower or taller/smaller hits runtime assertions sporadically.
The 2 assertions I have seen:
File: src/ui/canvas/gdi/BufferCanvas.cpp, Line 93
File: src/ui/canvas/gdi/BufferCanvas.cpp, Line 104

Both say "Expression: success".
Both can be Ignored.

Steps to reproduce the behavior
-------------------------------
Start XCSoar on a PC, get past the profile selection into "normal" mode.
Then resize the window.

Do you have any idea what may have caused this?
-----------------------------------------------
Maybe some race conditions in the code?
Unfortunately my debug tools on Windows are non-existent.

Do you have an idea how to solve the issue?
-------------------------------------------
BTW a small nit:
src/ui/canvas/gdi/BufferCanvas.cpp line 90 declares "bool success" but only if DEBUG is set.
It wouldn't compile if NDEBUG is set.

lordfolken, 2024-09-27:
I haven´t  observed this in quite some time. Do you think we can close @bomilkar ?

bomilkar, 2024-09-28:
You are right: I haven't seen these assertion/crash in a while.
I haven't been testing a lot on WIN64 lately.
We can always re-open if it appears again.

closed 2024-09-28 by bomilkar