XCSoar/XCSoar #603: crash with Runtime Assertion on WIN64

bomilkar, 2021-05-02:
XCSoar versions having and not having the problem
-------------------------------------------------
Information from the Credits page:
version: 7.6-PC
date: May 2 2021
git: ba1a6db0f7

Expected behavior
-----------------
Should not crash.

Actual behavior
---------------
When running for a while, either replay or live with moving location, it eventually hits a Runtime Assertion:
src/util/AllocatedGrid.hxx, Line 124
Expression: x < width

If you skip the assertion with "Ignore" it will crash.

Steps to reproduce the behavior
-------------------------------
Replay a log file for some time.

Do you have any idea what may have caused this?
-----------------------------------------------
GetPointerAt() gets called with unreasonable x and/or y coordinates.
Unfortunately my debug skils on Windows are non-existent, and on UNIX this part of the code isn't used.

Do you have an idea how to solve the issue?
-------------------------------------------
Hard to tell w/o a back trace or core dump.

lordfolken, 2023-09-22:
I believe this to be fixed? Can you confirm @bomilkar ?

bomilkar, 2023-09-23:
I haven't seen it in a while.
I'll re-open it should it reappear.

closed 2023-09-23 by bomilkar