XCSoar/XCSoar #604: Altitude not used when replaying an IGC file

Flying-Roger, 2021-05-02:
XCSoar versions having and not having the problem
-------------------------------------------------
Android XCSoar version 7.6 has the problem.
The problem was also there in previous versions.

Expected behavior
-----------------
When replaying an IGC file, I would expect that all elements related to the altitude are calculated based on the recorded altitude.

Current behavior
-------------------
Only the "GPS Altitude" is working, while the other altitude and variometer elements are frozen.
Due to this, it is not possible to really get experienced with all XCSoar functions by replaying IGC files.

Steps to reproduce the behavior
-------------------------------
Config 2 > Replay > load an IGC file > Start

Do you have any idea what may have caused this?
--------------------------------------------------------
No

Do you have an idea how to solve the issue?
-------------------------------------------------
Probably the bug can be fixed by copying the GPS altitude from the IGC file onto the "Baro Altitude".

MaxKellermann, 2021-08-23:
Can't reproduce. If I replay IGC files, I get valid altitudes and vario values.
Your bug report does not contain any sample IGC file.

closed 2021-08-23 by MaxKellermann

FlyingRoger, 2021-08-24:
Sorry for not providing a sample IGC.
Actually, I had the problem with any IGC, all recorded without barometric data.
Today, I tested today with version 7.15-Android dated Aug 19, 2021.
Good news = the problem is seemingly solved with this version: vario/altitude related data are now operating during a replay of my IGC files.

Many thanks to the developper who fixed the problem !

FlyingRoger, 2021-08-30:
Sorry for coming back on this issue, but the problem is actually not solved.
However, I found an interesting element:
- The problem temporarily disappears if, during the replay, I click on the "+10'" button.
Then the vario becomes effective and the AGL altitude becomes correct as long as the replay moves ahead in accelerated mode.
When it resumes at the specified speed (1x, 2x,...), the vario fails again and the AGL altitude is again wrong (negative).
[Sample_IGC_vario_problem.txt](https://github.com/XCSoar/XCSoar/files/7076479/Sample_IGC_vario_problem.txt)
[image]
[image]
[image]
Here attached a few snapshots to show this.
I also provide a short igc to reproduce the problem.