XCSoar/XCSoar #608: Untranslated InfoBoxes string / i18n / fr

scls19fr, 2021-05-02:
Hello,

the following infoboxes are not translated into french (XCSoar v7.6 Android) :

- Active Radio Frequency
- Contest distance
- Contest speed
- Standby Radio Frequency
- Thermal time

Oddly, 
- Contest distance (Distance On-Line Contest)
- Contest speed (Vitesse sur circuit "OLC") 
was translated in XCSoar-testing (which is currently numbered v7.2-Android)

Kind regards

Scumi, 2021-05-02:
> Oddly,
> 
>     * Contest distance (Distance On-Line Contest)
> 
>     * Contest speed (Vitesse sur circuit "OLC")
> 
> 
> was translated in XCSoar-testing (which is currently numbered v7.2-Android)

This is linked to #575, which happened after the version you mentioned.

scls19fr, 2021-05-04:
InfoBoxes layout are partially translates into french also. I can provide PR if someone point me to file to édit.

scls19fr, 2021-05-05:
I have found https://hosted.weblate.org/translate/xcsoar/translations/fr/

closed 2021-05-05 by scls19fr