XCSoar/XCSoar #609: InfoBox Distance custom waypoint

scls19fr, 2021-05-04:
Hello,

It could be a nice feature to have an InfoBox which could display distance to a custom waypoint.
So user could display both 
- Distance to next waypoint (nav)
- Distance to an other point (outside of nav)

Kind regards

MaxKellermann, 2021-05-04:
There's a distance home infobox already.

scls19fr, 2021-05-04:
Home is not a custom waypoint. I don't change home during flight

elgandoz, 2021-09-05:
@scls19fr You can use the ATC radial widget: it allows to "relocate" to any waypoint or landable, providing you radial degrees from the waypoint and the distance from it.
[image]

scls19fr, 2021-09-05:
Thanks @elgandoz this is an interesting feature. Unfortunately distance seems to be in nautical miles only. Being able to change to km will be great.

scls19fr, 2021-09-05:
For my use case having distance bigger than radial will be better